sqfmi/Watchy #121: No WiFi network created during setup

kylecarow, 2021-12-20:
Got this as a Christmas gift and I'm excited to tinker with it. When I set up WiFi connections earlier there were no issues, however, after fixing some out-of-the-box issues by uploading a new watch face, I can no longer set up WiFi connections. I can't seem to find a Watchy AP network whenever I try. Any help?

darianbjohnson, 2021-12-22:
I'm getting the same error. Would welcome any help!

btwebz, 2021-12-23:
I'm getting the exact same error. Would also welcome any help please! Tried hard-coding the WiFi connection details and it is still not working.

JanCBrammer, 2021-12-25:
Exact same issue for me.

btwebz, 2021-12-25:
Rolling back one version ended up working for me.

kylecarow, 2021-12-27:
> Rolling back one version ended up working for me.

Can confirm rolling back to 1.2.11 fixed the issue. Something about the changes to 1.2.12 goofs the WiFi network feature up. The newer version also garbles the error text when failing to connect, not sure why.

Pome-ro, 2021-12-28:
@kylecarow @btwebz  I'm experiencing this too, could you provide more detail on how you rolled back to 1.2.11?

kylecarow, 2021-12-28:
> @kylecarow @btwebz  I'm experiencing this too, could you provide more detail on how you rolled back to 1.2.11?

If you go to the Arduino library manager and search Watchy you should be able to select and install any version. Alternatively, you can edit the library itself (in your Arduino/libraries directory) to use your SSID and password.

sqfmi, 2021-12-29:
Fixed in v1.3.0. The lightsleep on busy wait power optimization caused the WiFi to sleep as well. Please re-open if issue persists.

closed 2021-12-29 by sqfmi