sqfmi/Watchy #123: Timelib expects 1970 offset, WatchyRTC is wanting 2000.

GuruSR, 2021-12-22:
typedef struct  { 
  uint8_t Second; 
  uint8_t Minute; 
  uint8_t Hour; 
  uint8_t Wday;   // day of week, sunday is day 1
  uint8_t Day;
  uint8_t Month; 
  uint8_t Year;   // offset from 1970; 
} 	tmElements_t, TimeElements, *tmElementsPtr_t;

Year is expected to be an offset from 1970, with current WatchyRTC set/read going to 2000, local time calculation may be off in some situations.  Suggest making the set/read add/subtract 30 from the PCF8563 and leave the DS3231 as is.

dezign999, 2021-12-22:
On my DS3231 I was in 2021 in the before times. Now I'm living in 2051, I'm a Bladerunner now B)

GuruSR, 2021-12-22:
If issue 124 is met and the 1900 century flag is set on the PCF8563, then both of the RTCs can lose the year offset changes, as they'll be running expecting 1900 year offset.

closed 2022-01-03 by GuruSR