sqfmi/Watchy #128: Display is stuck at 00:00

neijka, 2021-12-25:
The display is stuck at 00:00. 
It happened after being stuck on Bluetooth. I uploaded a watch face and new display got stuck at 00:00. 
I cannot manage to open the case to unplug the battery. How do you open the case (to unplug the battery) ? Any other alternatives ? Advice ?

sqfmi, 2021-12-27:
You should still be able to upload a new sketch again via USB. You can also open the case by using a small tool to push on the 2 snaps at the bottom.

neijka, 2021-12-31:
Thank you. I opened it. Menu is working. I can upload sketches. But display is still stuck at 00:00

sqfmi, 2021-12-31:
Have you tried that latest update v1.3.1 and sync NTP to get the latest time (after configuring wifi)?

stuartpb, 2022-03-30:
I'm on 1.3.3, and I'm also seeing this problem - the device will connect to WiFi correctly and sync NTP (displaying the correct time on the confirmation screen), but the main watch face still displays 00:00 - even setting the time manually won't fix the display. I've tried running `pio run -t erase && pio run -t upload`, and the problem persists.

stuartpb, 2022-03-31:
Running `pio run -t upload` again after doing the NTP sync makes the screen briefly display the correct time, but the next screen refresh sets the display back to 00:00 and "SUNDAY 00 ERR 2000".

donpdonp, 2022-04-21:
I am also having this problem. Setting the time manually has no effect. Sync to NTP succeeds but has no effect. Doing a pio run -t upload for some reason sets the time correctly, but only for a moment. the screen refreshes and back to 00:00.

neaket, 2022-05-14:
I just received my Watchy.  I had the same issue as above.  

It looks like I fixed the issue by adding a call to `rtc_pcf.getDateTime()` in WatchyRTC::read.

https://github.com/sqfmi/Watchy/blob/3e3a8859a94755cc0b7d2de5f189a3de7ccb62e7/src/WatchyRTC.cpp#L46-L60

Updated:
```
void WatchyRTC::read(tmElements_t &tm) {
  if (rtcType == DS3231) {
    rtc_ds.read(tm);
  } else {
    rtc_pcf.getDateTime();
    tm.Year  = y2kYearToTm(rtc_pcf.getYear());
    tm.Month = rtc_pcf.getMonth();
    tm.Day   = rtc_pcf.getDay();
    tm.Wday =
        rtc_pcf.getWeekday() + 1; // TimeLib & DS3231 has Wday range of 1-7, but
                                  // PCF8563 stores day of week in 0-6 range
    tm.Hour   = rtc_pcf.getHour();
    tm.Minute = rtc_pcf.getMinute();
    tm.Second = rtc_pcf.getSecond();
  }
}
```

donpdonp, 2022-05-14:
@neaket that did it! thank you! now my Wachy is a watch again. This one-liner needs to go into the Watchy package ASAP! Also to package maintainer(s): tag v1.4.1 still has config.h with 75:#define WATCHY_LIB_VER "1.4.0"

closed 2022-07-10 by sqfmi

portothree, 2022-09-23:
Hey, I'm facing a issue building master since this commit.
```
.pio/libdeps/esp32dev/Watchy/src/WatchyRTC.cpp: In member function 'void WatchyRTC::read(tmElements_t&)':
.pio/libdeps/esp32dev/Watchy/src/WatchyRTC.cpp:50:13: error: 'class Rtc_Pcf8563' has no member named 'getDateTime'
     rtc_pcf.getDateTime();
```

Looking at the RTC library Watchy uses (https://github.com/orbitalair/Rtc_Pcf8563), it does not seem like there's a `getDateTime` method. https://github.com/orbitalair/Rtc_Pcf8563/blob/92e51faa40b6859bce6a4b468da6a1a551e4b6ab/keywords.txt

PR #168 (2022-07-10): PCF8563: Call rtc_pcf.getDateTime in WatchyRTC::read
Fixes #128 as noted by @neaket
commit 37ea6135 (2022-07-10): PCF8563: Call rtc_pcf.getDateTime in WatchyRTC::read

Fixes #128 as noted by @neaket