sqfmi/Watchy #134: Can watchy be reset by holding buttons?

rldleblanc, 2021-12-30:
Sometimes the watch locks up and I have to take it apart to reset the power on it. Is there some shortcut I'm missing to reset it like a phone where you hold down one or more buttons for some number of seconds? If not is it possible to send a reset to the Arduino? I'm really new to Arduino so some push in the right direction I may be able to contribute to the project.

sqfmi, 2022-01-20:
The only method of resetting in the current revision is to remove power (usb or battery).

closed 2022-01-20 by sqfmi

rldleblanc, 2022-01-20:
So even though the Ardunio IDE can reset the watch automatically, there is no way to listen for a long button press and do the same thing?

sqfmi, 2022-01-21:
That is correct. You can do a soft reset using ```ESP.restart()``` but won't work if the code locks up.