sqfmi/Watchy #136: `watchySettings` struct is undocumented

tetsuo-cpp, 2022-01-01:
Hey there, I was following the instructions [here](https://watchy.sqfmi.com/docs/create-watchface) and noticed that `Watchy` no longer has a default constructor and instead takes a `watchySettings`. I ended up copying across a `settings.h` from one of the preset watch faces.

I'd be happy to make a PR for https://github.com/sqfmi/watchy-docs, but I figured I'd check with you in case you've already got something in-progress.

sqfmi, 2022-01-01:
Thanks @tetsuo-cpp! A PR would be great, we're a little behind on documenting some of the recent changes.

JollyWizard, 2023-01-09:
I believe this is going to create a problem with the current gallery examples.

Tried using the Kitty watch from the gallery (the only one my daughter desperately wanted), and it depends on the default constructor, but does not include any versioning info, so I couldn't figure out what to peg it to to work. Having other slow and compilation problems with both Arduino and platformio, so it's hard to narrow it all down.  The removal of default constructor definitely put a break in the code base, though.

Please reinstate a default constructor or do a review of all gallery examples and post migration guidance.

Thanks.

meisterluk, 2024-09-18:
I just want to add an explicit fix for people not that familiar with programming. If you want to get the example Rand_Pics running, follow the following instructions:

watchySettings has the following structure (acc. to [Watchy.h](https://github.com/sqfmi/Watchy/blob/9f3accdb076b567c2e34b0347f292effcde04a6f/src/Watchy.h#L48)):

```C
typedef struct watchySettings {
  // Weather Settings
  String cityID;
  String lat;
  String lon;
  String weatherAPIKey;
  String weatherURL;
  String weatherUnit;
  String weatherLang;
  int8_t weatherUpdateInterval;
  // NTP Settings
  String ntpServer;
  int gmtOffset;
  //
  bool vibrateOClock;
} watchySettings;
```

So you need to instantiate it:

```C
watchySettings mySettings{
    .cityID = "2778067",
    .lat = "47.066669",
    .lon = "15.45",
    .weatherAPIKey = "HERE-I-INSERTED-MY-API-KEY",
    .weatherURL = "http://api.openweathermap.org/data/2.5/weather?lat={lat}&lon={lon}&lang={lang}&units={units}&appid={apiKey}",
    .weatherUnit = "metric",
    .weatherLang = "en",
    .weatherUpdateInterval = 30, //must be greater than 5, measured in minutes
    .ntpServer = "pool.ntp.org",
    .gmtOffset = 0, //New York is UTC -5 EST, -4 EDT, will be overwritten by weather data
    .vibrateOClock = true,
};
```

The API key can easily be requested if you are registered at [openweathermap.org](https://openweathermap.org/). I did not try it out with an *invalid* API key. I inserted these lines in `Rand_Pics.cpp` just [before the line](https://github.com/wolfdtx/RandPics/blob/bbf4714948d831c3e7ce987741c4a3758305f815/Rand_Pics.cpp#L4) ``//Ratfink Artsheet``.

And then I replaced this line …

```C
RandPics::RandPics(){} //constructor
```

… with the following line:

```C
RandPics::RandPics(): Watchy(mySettings) {} //constructor
``` 

Finally, I had to rename all four files with filename `Rand_pics.h` to `Rand_Pics.h`.

Afterwards I was able to click the “Compile” button in Arduino IDE and did not get an error anymore.