sqfmi/Watchy #140: Bluetooch protocol and Gadgetbridge support

shtrom, 2022-01-08:
I know this is a stretch, but I want to to tie up #90 and #102, and create a single location for discussion of ideas towards this goal.

One of the advantages of smart watches is the ability to connect to a paired mobile, over Bluetooth or BLE, and get info pushed/pulled from it, such as notifications. While this can also be used to provide additional Internet-based info to the watch, such as weather or time sync, Watchy currently already does it, so while it would be nice to have, I don't think that's a priority (even though it could be done in the long run).

https://github.com/Freeyourgadget/Gadgetbridge/ is an open source project that aims at supporting a number of watches, proprietary or not. This means most of the mobile-side handling could already be taken care off.

The main thing we'd need to focus on would be the Bluetooth protocol to be able to make both devices communicate. We could then implement it within Gadgetbridge at the same time as in the Watchy code. An alternative would be to just use an already supported protocol (my heart would got to the Pebble's for that, but there's nothing forcing our hand here).

shtrom, 2022-01-08:
Actually, @thelastproject already opened a ticket there: https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/2326

rastagraffix, 2022-01-09:
I'm all in on this.  Now that I'm able to push firmware to my Watchy my first major goal is to get it syncing time from my iPhone over BLE.  I don't consider SNTP a viable solution for a couple of reasons: 1) I'm often out of range of the configured Wifi network, and 2) BLE consumes a lot less current when the radios are active.

Ditto for weather data.

My first stumbling block is figuring out how to get BLE to pair with the iPhone -- iOS won't offer its Time service to an unpaired BLE device.  When I have that solved it should be easy to have the watch poll the phone for the time every hour or so.

And, as with my Pebble, it would be nice if I could selectively push notifications to the watch, but that assumes we can leave the Watchy BLE radio powered up to receive the notifications without killing the battery.

cafehaine, 2022-08-17:
Hi!

I would love to help develop a firmware/phone app to have notification/music control working on the Watchy, but I'm not sure how we would tackle this.

I think creating a protocol from scratch might be the simplest solution.

Disclamer: I've never dealt with bluetooth/BLE before, so the following might not match perfectly the concept of bluetooth services.

## My protocol idea

It's just how I would implement something like this, it's probably imperfect so please, tell me if something looks wrong to you !

All the following services would communicate with the following packet structure :

### First packet

| Total length of message | Date and time (to update the RTC) |
|-------------------------|-----------------------------------|
| unsigned 32bit int      | signed 64 bit unix timestamp      |

### Following packets

| Message contents |
|------------------|
| Up to 512 bytes  |

### Data Types

- `GUID`: 16 bytes
- `char`: 1 byte
- `string`: 2 bytes length + n bytes data
- `image`: raw data (5000 bytes for 200x200, 128 for 32x32)
- `int`: 2 bytes signed

### Messages (each one a separate bluetooth service?)

- Notification changed
  - Notification GUID: GUID
  - Notification title: string
  - Notification text: string
  - Notification icon: image (32x32 1bpp)
- Notification removed
  - Notification GUID: GUID
- Media playing changed
  - Media title: string
  - Media album: string
  - Media cover: image (200x200 1bpp)
- Media stopped playing
  - Empty
- Phone call incoming
  - Contact name: string
  - Phone number: string
  - Contact illustration: image (200x200 1bpp)
- Phone call started
  - Contact name: string
  - Phone number: string
  - Contact illustration: image (200x200 1bpp)
- Phone call ended
  - Empty
- Set wallpaper
  - Wallpaper: image (200x200 1Bpp)
- Set watchface
  - Watchface GUID: GUID
- Update weather information
  - Temperature: int
  - Temperature unit: char (M or I)
  - Weather: string ('main' field of openweathermap API)

Crsarmv7l, 2022-09-14:
https://github.com/sharandac/My-TTGO-Watch

Uses an ESP32 and pretends to be a Bangle.js for gadgetbridge support. While you could certainly write your own code from scratch, I would bet it might be easier to adapt his code.

celevra, 2024-02-24:
Any news on that?
Still no way to get Phone notifications?

fbiego, 2025-01-31:
Here is a suggestion for BLE
https://github.com/fbiego/chronos-esp32
- Time
- Notifications
- Weather 
- Controls (Music, Find Phone, Camera)
- Phone Battery Status
- Google Map Navigation
- Contacts & QR codes