sqfmi/Watchy #143: 1.4.0 build broken: missing pins_arduino.h

rastagraffix, 2022-01-21:
In 1.4.0 the soc pin definitions were moved to pins_arduino.h.  However that file hasn't been committed to the repo, therefore the build is broken.

sqfmi, 2022-01-21:
1.4.0 hasn't been tagged yet, it will be when ESP32 Arduino core gets updated

rastagraffix, 2022-01-21:
Any chance of getting a copy of that file so I can build in the interim?

rastagraffix, 2022-01-21:
I.e. commit it into src/ on the dev branch until the esp32 core gets updated, then rm it.

sqfmi, 2022-01-21:
Sounds good, added in dev branch https://github.com/sqfmi/Watchy/commit/da92b9d4a1aad57174a9be01bfc5bda148161f07

rastagraffix, 2022-01-21:
Nope, still broken. The hardware version macro isn't being defined, so this block of #ifs never matches:
```
#if     defined (ARDUINO_WATCHY_V10)
    static const uint8_t UP_BTN_PIN   = 32;
    static const uint8_t BATT_ADC_PIN = 33;
    #define UP_BTN_MASK  GPIO_SEL_32
    #define RTC_TYPE 1 //DS3231
#elif defined (ARDUINO_WATCHY_V15)
    static const uint8_t UP_BTN_PIN   = 32;
    static const uint8_t BATT_ADC_PIN = 35;
    #define UP_BTN_MASK  GPIO_SEL_32
    #define RTC_TYPE 2 //PCF8563
#elif defined (ARDUINO_WATCHY_V20)
    static const uint8_t UP_BTN_PIN   = 35;
    static const uint8_t BATT_ADC_PIN = 34;
    #define UP_BTN_MASK  GPIO_SEL_35
    #define RTC_TYPE 2 //PCF8563
#endif
```

leading to
```
In file included from /Users/lyndon/Documents/Arduino/libraries/Watchy/src/WatchyRTC.h:6,
                 from /Users/lyndon/Documents/Arduino/libraries/Watchy/src/Watchy.h:14,
                 from /Users/lyndon/Documents/Arduino/libraries/Watchy/src/Watchy.cpp:1:
/Users/lyndon/Documents/Arduino/libraries/Watchy/src/Watchy.cpp: In member function 'void Watchy::deepSleep()':
/Users/lyndon/Documents/Arduino/libraries/Watchy/src/config.h:37:51: error: 'UP_BTN_MASK' was not declared in this scope
 #define BTN_PIN_MASK  MENU_BTN_MASK|BACK_BTN_MASK|UP_BTN_MASK|DOWN_BTN_MASK
                                                   ^~~~~~~~~~~
   .
   .
   .
```

I think I'll just wait for the upstream esp32 repo to sync up :-)  Thanks for trying.

sqfmi, 2022-01-23:
You could define it at the top based on your HW version, but yeah we'll revert it for now and wait for the upstream

rastagraffix, 2022-01-30:
Do you have any idea when the upstream is going to push this into an actual release that the Arduino IDE can update to? Not being able to build is really frustrating. 

If that's going to take a while then these changes need to be moved to their own branch and reverted from master and dev until that happens. Right now you're forcing everyone to install and track the bleeding edge espressif arduino repo. We shouldn't have to do that.

HyrumF, 2022-04-16:
Noon here! Is my problem related to your issue? 

I just got my watch, followed the instructions, and tried to upload some of the Watchy examples.  They all give errors. 
I'm on Arduino 1.8.19 and  Watchy 1.3.3

```
WARNING: library Rtc_Pcf8563 claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
C:\Users\hkfai\Documents\Arduino\libraries\Watchy\examples\WatchFaces\Tetris\Watchy_Tetris.cpp: In constructor 'WatchyTetris::WatchyTetris()':
Watchy_Tetris.cpp:5:28: error: no matching function for call to 'Watchy::Watchy()'
 WatchyTetris::WatchyTetris(){} //constructor
                            ^
In file included from C:\Users\hkfai\Documents\Arduino\libraries\Watchy\examples\WatchFaces\Tetris\Watchy_Tetris.h:4,
                 from C:\Users\hkfai\Documents\Arduino\libraries\Watchy\examples\WatchFaces\Tetris\Watchy_Tetris.cpp:1:
C:\Users\hkfai\Documents\Arduino\libraries\Watchy\src/Watchy.h:47:18: note: candidate: 'Watchy::Watchy(const watchySettings&)'
         explicit Watchy(const watchySettings& s) : settings(s){} //constructor
                  ^~~~~~
C:\Users\hkfai\Documents\Arduino\libraries\Watchy\src/Watchy.h:47:18: note:   candidate expects 1 argument, 0 provided
C:\Users\hkfai\Documents\Arduino\libraries\Watchy\src/Watchy.h:40:7: note: candidate: 'Watchy::Watchy(const Watchy&)'
 class Watchy {
       ^~~~~~
C:\Users\hkfai\Documents\Arduino\libraries\Watchy\src/Watchy.h:40:7: note:   candidate expects 1 argument, 0 provided
C:\Users\hkfai\Documents\Arduino\libraries\Watchy\src/Watchy.h:40:7: note: candidate: 'Watchy::Watchy(Watchy&&)'
C:\Users\hkfai\Documents\Arduino\libraries\Watchy\src/Watchy.h:40:7: note:   candidate expects 1 argument, 0 provided
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\hkfai\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
no matching function for call to 'Watchy::Watchy()'

```

synthead, 2022-05-12:
It looks like https://github.com/sqfmi/Watchy/commit/da92b9d4a1aad57174a9be01bfc5bda148161f07 was shipped with v1.4.0: https://github.com/sqfmi/Watchy/compare/v1.4.0...master

In addition, version v1.4.1 is available in Arduino now, which also includes the fixes:

[image]

Can this issue be closed?  I think @rastagraffix's concerns have been addressed.

rastagraffix, 2022-05-13:
Yes, if the build is fixed (I won't be able to check mine until the weekend) then there's no reason to keep this issue open.

sqfmi, 2022-12-29:
This should be resolved in the latest release, please re-open if the issue persists. Thanks!

closed 2022-12-29 by sqfmi