sqfmi/Watchy #146: pio run is giving compile errors about most defines.

halkeye, 2022-02-18:
I feel like I have something outdated or misinstalled or something. I'm trying out pio for watchy for the first time.

My guess is that something isn't being included properly?

```
PS C:\Users\Gavin Mogan\Documents\my_new_watchy_face_project> pio run
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.5.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
Converting Pokemon.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep+, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Watchy> 1.4.0+sha.cd17d20
|   |-- <Adafruit GFX Library> 1.10.13
|   |   |-- <Adafruit BusIO> 1.11.1
|   |   |   |-- <Wire> 1.0.1
|   |   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <Arduino_JSON> 0.1.0
|   |-- <DS3232RTC> 0.0.0-alpha+sha.261ca7d0e6
|   |   |-- <Time> 1.6.1
|   |   |-- <Wire> 1.0.1
|   |-- <NTPClient> 3.1.0
|   |-- <Rtc_Pcf8563> 1.0.3+sha.92e51fa
|   |   |-- <Wire> 1.0.1
|   |-- <GxEPD2> 1.4.5+sha.31e52bd
|   |   |-- <SPI> 1.0
|   |   |-- <Adafruit GFX Library> 1.10.13
|   |   |   |-- <Adafruit BusIO> 1.11.1
|   |   |   |   |-- <Wire> 1.0.1
|   |   |   |   |-- <SPI> 1.0
|   |   |   |-- <Wire> 1.0.1
|   |   |   |-- <SPI> 1.0
|   |-- <ESP32 BLE Arduino> 1.0.1
|   |-- <WifiManager> 2.0.3-alpha+sha.9af7253
|   |   |-- <WiFi> 1.0
|   |   |-- <WebServer> 1.0
|   |   |   |-- <WiFi> 1.0
|   |   |   |-- <FS> 1.0
|   |   |-- <DNSServer> 1.1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <WiFi> 1.0
|   |-- <WebServer> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <FS> 1.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <WiFi> 1.0
|   |-- <HTTPClient> 1.2
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|   |-- <Time> 1.6.1
|-- <WifiManager> 2.0.3-alpha+sha.9af7253
|   |-- <WiFi> 1.0
[... 114 lines cut ...]
     pinMode(DOWN_BTN_PIN, INPUT);
             ^
.pio/libdeps/esp32dev/Watchy/src/Watchy.cpp:338:13: error: 'UP_BTN_PIN' was not declared in this scope
     pinMode(UP_BTN_PIN, INPUT);
             ^
.pio/libdeps/esp32dev/Watchy/src/Watchy.cpp:339:13: error: 'MENU_BTN_PIN' was not declared in this scope
     pinMode(MENU_BTN_PIN, INPUT);
             ^
.pio/libdeps/esp32dev/Watchy/src/Watchy.cpp:340:13: error: 'BACK_BTN_PIN' was not declared in this scope
     pinMode(BACK_BTN_PIN, INPUT);
Compiling .pio\build\esp32dev\FrameworkArduino\esp32-hal-i2c.c.o
             ^
.pio/libdeps/esp32dev/Watchy/src/Watchy.cpp: In member function 'void Watchy::showAccelerometer()':
.pio/libdeps/esp32dev/Watchy/src/Watchy.cpp:491:13: error: 'BACK_BTN_PIN' was not declared in this scope
     pinMode(BACK_BTN_PIN, INPUT);
             ^
.pio/libdeps/esp32dev/Watchy/src/Watchy.cpp: In member function 'float Watchy::getBatteryVoltage()':
.pio/libdeps/esp32dev/Watchy/src/Watchy.cpp:615:37: error: 'BATT_ADC_PIN' was not declared in this scope
         return analogReadMilliVolts(BATT_ADC_PIN) / 1000.0f * 2.0f; // Battery voltage goes through a 1/2 divider.
                                     ^
.pio/libdeps/esp32dev/Watchy/src/Watchy.cpp:617:37: error: 'BATT_ADC_PIN' was not declared in this scope
         return analogReadMilliVolts(BATT_ADC_PIN) / 1000.0f * 2.0f;
                                     ^
*** [.pio\build\esp32dev\lib59e\Watchy\Watchy.cpp.o] Error 1
```

cscott, 2022-03-01:
Yeah, I'm seeing this too when I compile the example according to the instructions at https://watchy.sqfmi.com/docs/getting-started#platformio-setup

halkeye, 2022-03-01:
Its apparently related to https://github.com/sqfmi/Watchy/issues/143? I havn't had a chance to try it again

cscott, 2022-03-01:
Quite possibly related to #143.  Adding a `build_flags` clause and changing the `lib_deps` to pin to the commit just before ee57707 fixes the problem.  (Use `-DARDUINO_WATCHY_V10` if you have an older watchy.)
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
build_flags = -DARDUINO_WATCHY_V15
lib_deps =
    https://github.com/sqfmi/Watchy.git#da92b9d4a1aad57174a9be01bfc5bda148161f07
lib_ldf_mode = deep+
board_build.partitions = min_spiffs.csv
```

sqfmi, 2022-12-29:
This should be resolved in the latest release, please re-open if the issue persists. Thanks!

closed 2022-12-29 by sqfmi