sqfmi/Watchy #147: Failed to connect to ESP32

jeffsanicola, 2022-02-19:
Hello!

When attempting to upload a sketch to my Watchy I'm receiving an error `A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header`. I was able to upload sketches previously, however recently I've been unable to and getting this message consistently.

Full log output is:
```
WARNING: library DS3232RTC claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
WARNING: library Rtc_Pcf8563 claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
Sketch uses 1699093 bytes (86%) of program storage space. Maximum is 1966080 bytes.
Global variables use 52244 bytes (15%) of dynamic memory, leaving 275436 bytes for local variables. Maximum is 327680 bytes.
esptool.py v3.1
Serial port COM3
Connecting........_____....._____....._____....._____....._____....._____.....____An error occurred while uploading the sketch
_

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
```

After the message is displayed I do see the unit restart as if the reset pin was being applied (although that's not stated in the logs anywhere).

Here are the version numbers that I could location based on the [Libraries Used](https://watchy.sqfmi.com/docs/libs) page
Arduino IDE v1.8.19 on Windows 10 Pro
Espressif Systems esp v2.0.2
Adruino_JSON v0.1.0
Adafruit GFX v1.10.13
DS3232RTC v1.3.0
GxEPD2 v1.4.5
Watchy v1.3.3
WiFiManager v2.0.9-beta

I've tried the simple things like a different USB cable/port, disconnect/reconnect the battery, try to push without the battery connected, and update outside of the case (I have the aluminum one) but all result in the same issue. Everything else seems to work - I'm able to sync NTP and weather details from WiFi, the vibration motor and motion sensor all seem to be working fine as well.

Does anybody know of a way to resolve this?

sqfmi, 2022-03-03:
Do you have the [drivers](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers) installed and confirm it is COM3 (you can check in device manager)? Also, do you have another computer you could try to see if the serial port is detected (Linux/OSX)?

jeffsanicola, 2022-03-05:
@sqfmi - yes, I updated to the latest drivers and do see the device in Device Manager and I also see the device listed in Fedora Linux when running `lsusb`.

```bash
[jeff@localhost ~]$ lsusb | grep Silicon
Bus 001 Device 040: ID 10c4:ea60 Silicon Labs CP210x UART Bridge
```

I believe the last firmware I was able to install was v1.3.2 and the hardware revision I have is the one that was affected by issue [106](https://github.com/sqfmi/Watchy/issues/106).

tbjers, 2022-03-22:
I also have the same problem, I think after I tried to use the beta feature via BlueTooth? Or it could have been there all along.

loonylion, 2022-03-22:
> I also have the same problem, I think after I tried to use the beta feature via BlueTooth? Or it could have been there all along.

likely been there all along, I've never used the bluetooth feature.

searsaw, 2022-03-27:
I am also having this issue. I was able to do a Bluetooth update of the watch face to put Basic on it, but now I'm having the issue in #128 which unfortunately means I can't update to another watch face because I can't get the menu to come up. :disappointed: I have tried doing the update using Arduino IDE, `pio`, and PlatformIO IDE in VS Code. Also, when it tries to update, it fails. However, the watch screen resets itself when it fails. So it seems like it's trying.

@tbjers I had issues with Bluetooth until I found out that Chrome doesn't turn on the BLE functionality by default on Linux. it needs to be turned on in `about::flags`. However, it doesn't work because it tries to pair to the watch and closes the connection before the firmware can be passed. I had to update the built firmware from the GitHub repo using the WiFi setup screen.

walex53, 2022-03-28:
@jeffsanicola Just do, what IrishJourno wrote in #23 : Erase the esp32 flash with the help of a Raspberry Pi and then upload the new code from your Arduino IDE. Sounds weird, but it works. And he wrote that more than 1 year ago and the problem still persists!

On my Desktop PC (Linux Debian Buster, Kernel 4.19.0-20-amd64) I type: "python3 -m esptool erase_flash" and I get:
esptool.py v3.3
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting......................................
/dev/ttyUSB0 failed to connect: Failed to connect to Espressif device: No serial data received.

But when I connect the Watchy to my Raspberry Pi and type the same command, I get:
esptool.py v3.3
Found 2 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 4c:75:25:ab:6e:28
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 5.6s
Hard resetting via RTS pin...

After that you plug the Watchy back into your PC and the Arduino IDE works fine and uploads your code to the device.
Please, could anybody with a Logic Analyzer find out the timing differences between the Raspberry Pi and a Desktop PC?
I surely will not solder a big fat capacitor on to my Watchy board, to "bend" any timing, that goes too fast :-)

@sqfmi Could you please update the chapter "Arduino Setup" on https://watchy.sqfmi.com/docs/getting-started ?
If you follow these instructions and install all the latest libraries, you will get a lot of compiler error messages in the Arduino IDE.
The use of Watchy Ver.1.1.12 and DS3232RTC Ver.1.3.0 worked for me. If you install the latest versions, it doesn't compile. 
All the other libraries and the esp32 platform I use the latest versions.

searsaw, 2022-03-28:
@walex53 Using an rpi to clear the flash works. I was then able to upload a watch face via my Linux machine afterward. However, I have to erase the flash each time I want to upload a new watch face. Really is a bummer.

walex53, 2022-03-29:
UDATE

On my notebook the Arduino IDE runs flawlessly! It compiles, erases the flash memory and uploads the new code.
Notebook: Debian Buster, Kernel 4.19.0-16-amd64
Arduino IDE 1.8.19, esptool.py v3.1 from the esp32 package

Even on the command line the esptool.py v2.5.1, which I installed via the apt packet manager, runs without problems.

I still have no clue, why it does not run on my Desktop PC.

orlo11, 2022-03-29:
I have another connection error:
I used 2 different machines (both running Debian Bookworm, Kernel 5.16.14-1 amd64) and 3 different cables that work fine with my harddisks and my phone and of course all available usb-connectors, the driver cp210x is installed (via modprobe). I get always
`Mar 29 22:08:19 horsetower kernel: [ 3852.892865] usb 2-2: new full-speed USB device number 2 using xhci_hcd
Mar 29 22:08:20 horsetower kernel: [ 3853.029007] usb 2-2: device descriptor read/64, error -71
Mar 29 22:08:20 horsetower kernel: [ 3853.272961] usb 2-2: device descriptor read/64, error -71
Mar 29 22:08:20 horsetower kernel: [ 3853.508810] usb 2-2: new full-speed USB device number 3 using xhci_hcd
Mar 29 22:08:20 horsetower kernel: [ 3853.644947] usb 2-2: device descriptor read/64, error -71
Mar 29 22:08:20 horsetower kernel: [ 3853.888962] usb 2-2: device descriptor read/64, error -71
Mar 29 22:08:21 horsetower kernel: [ 3853.997498] usb usb2-port2: attempt power cycle
Mar 29 22:08:45 horsetower kernel: [ 3877.920750] usb 2-2: new full-speed USB device number 5 using xhci_hcd
Mar 29 22:08:45 horsetower kernel: [ 3878.060888] usb 2-2: device descriptor read/64, error -71
Mar 29 22:08:45 horsetower kernel: [ 3878.304839] usb 2-2: device descriptor read/64, error -71
Mar 29 22:08:45 horsetower kernel: [ 3878.540756] usb 2-2: new full-speed USB device number 6 using xhci_hcd
Mar 29 22:08:45 horsetower kernel: [ 3878.676850] usb 2-2: device descriptor read/64, error -71
Mar 29 22:08:46 horsetower kernel: [ 3878.920836] usb 2-2: device descriptor read/64, error -71
Mar 29 22:08:46 horsetower kernel: [ 3879.029797] usb usb2-port2: attempt power cycle`
in /var/log/syslog.
Does anyone have a suggestion, how to solve it. I cant upload any firmware.
Additional the watch becomes hot near the micro-usb connector, so that the e-ink-screen looses the ink, but it returns after draw the plug when the watch cools down.
Is there maybe the hardwarr broken?
Thanks for your answers,
orlo11.

walex53, 2022-03-29:
@orlo11 >watch becomes hot near the micro-usb connector
That does not sound good! USB can source 500 mA, that's enough to get hot!
Have a look at the backside of the Watchy board, where the little round buzzer motor is stick to the board with a double side adhesive tape. In my Watchy the solder pins of the motor are very close to the pins of an IC. Maybe there is a short cut.

r8tt8, 2022-03-30:
> UDATE
>
> On my notebook the Arduino IDE runs flawlessly! It compiles, erases the flash memory and uploads the new code. Notebook: Debian Buster, Kernel 4.19.0-16-amd64 Arduino IDE 1.8.19, esptool.py [...]

My problem was the same, regardless of linux/windows.
If i use my desktop with arch, it does NOT work: `Failed to connect to ESP32`.
If I use my laptop with both windows/arch, it works.

orlo11, 2022-03-31:
@walex53: Thanks for your suggestion. It is the CP2104 that gets very hot, whenever I connect the USB. so I will return the device and hope to get a new one soon.

Fell, 2022-04-07:
I have the same issue.

First I tried the Arduino IDE on my Windows machine, which couldn't compile because ALARM_2 was undefined, among other problems.

Then, I set up PlatformIO on Windows in Visual Studio Code which compiled (hooray) but gave me `Timed out waiting for packet header` every time. I installed the drivers, restarted everything, it just doesn't connect.

Then, I tried flashing directly with esptool, same problem.

Then, I hooked it up to my Raspberry Pi. It does recognize it as an ESP-PICO-D4, but also ultimately fails with `Timed out waiting for packet header`

I attempted to to the OTA Bluetooth flashing, and it just says `Waiting for upload` followed by `disconnected`. 

I tried all of this with both the battery attached and not attached, and it doesn't work.

I tried 3 different Micro USB cables from different manufacturers and different lengths. 

Is there anything else I can do?

walex53, 2022-04-07:
@Fell 
For the Compile Errors: Don't use the library DS3232RTC Version V2.0.0 ! Use Version 1.3.0 instead. See https://github.com/JChristensen/DS3232RTC , what the author of this library writes about changes. You can select the library version in the library manager of your Arduino IDE.

For the "can't upload" errors: It's not for the reason of a bad USB cable.
On the Raspberry Pi don't use the esptool that you installed by the apt package manager. Just copy the file esptool.py from the folder, that was created, when you installed the esp32 package in the Arduino IDE. On my Linux machine it is in: ".arduino15/packages/esp32/tools/esptool.py/3.1.0/esptool.py" (I hope, on a Windows machine the path is similar) and use it like  "python3 esptool.py flash_erase". Make shure to use the copied file "esptool.py" and not the one from the Raspberry OS Distribution.

After that plug the Watchy back to the machine with the Arduino IDE. It's only the flash_erase, that won't work on the Arduino IDE. The uploads works fine.

And did you try all of that from a Notebook? Maybe everything works fine. Good Luck!!!

Fell, 2022-04-07:
@walex53 Thank you so much! You just solved all my isses in one post.

Downgrading DS3232RTC from 2.0 to 1.3 fixed the compile error.

I then went and tried to copy the esptool.py from my Windows machine, but it turned out to be a esptool.exe. I just downloaded the [latest source code](https://github.com/espressif/esptool/releases/tag/v3.3) which contains the esptool.py as well. 

Executing `python3 esptool.py erase_flash` then actually managed to talk to the watch!

After doing that, I was able to flash another watchface using the ArduinoIDE.

walex53, 2022-04-07:
@Fell 
> Edit: So yeah, that works only once. After that, I have to hook it up to my Raspberry to erase the flash. Very strange. I'll try my laptop, but it's a windows machine too so I don't think it will work.

Yes, it only works once and you have to plug it over every time. It's the flash_erase, that doesn't work from a PC, but it does work from a Raspberry Pi and from a Notebook. For me, I made a workaround --> see picture. I soldered together some relays and USB jacks and glued it on to a ten year old Raspberry Pi I had laying around. A little script runs on it to automate the switching over to the Raspberry Pi and back to my Desktop PC. Yes, it's technical overkill and it's totally sick, but I like it. ;-)

But, honestly, isn't there any of the Watchy developers, who can fix this ?

[image]

loonylion, 2022-04-08:
erase_flash works on my framework laptop running funtoo linux. I have to do this before my desktop (MX Linux) can write the new image using arduino IDE.

Fell, 2022-04-09:
> @Fell
>
>
> Yes, it only works once and you have to plug it over every time. It's the flash_erase, that doesn't work from a PC, but it does work from a Raspberry Pi and from a Notebook. For me, I made a [...]

**Workarounds**
Wow, this is impressive.

I've been using a Raspberry on my desk via SSH, which is a bit of a hassle but bearable.

I'm thinking of buying a Raspberry Pi Zero and using it as an adaper.

**Back to the Problem**
At first I thought it's related to Windows, but it seems like the problem is related to desktop hardware. I tested my office PC with no luck. I have heard different things like:

Rumour 1: There should be an extra capacitor on the  GPIO_0 pin to help pulling it LOW. I don't think I could do that, it's way too tiny.

Rumour 2: Desktop PCs are supplying "too much" power which prevents pulling the pin low. If this is the case, maybe a very weak resistor could help? I have no idea, I'm a software guy.

Thought 3: After erasing the flash, flashing works with any PC, right? So as long as there is no code on the watch, it is flashable. Which leads me to believe that there might be a software problem with the bootloader on the watch, preventing it to go into flash mode under certain circumstances.

I don't know what's really going on, but maybe these thoughts spark the right idea to someone who knows.

Flynn-Mandrake, 2022-04-10:
That works! Neither the USB 3.0 nor the USB 2.0 Port on my desktop PC work, but if I plug my cable into the USB hub integrated into my monitor stand (which is itself plugged into another USB hub plugged into a USB 3.0 port in the back of my PC), it works flawlessly. This goes up there on the list of the weirdest issues I've ever encountered.

mniehus, 2022-04-10:
I am having the same 'timing' issue with my new watchy,  and could not connect on my ubuntu 20.04 be it via USB port 3.0 or 2.0. erase flash failed to communicate with ESP32. thanks for suggestions in this thread. the usb hub will be the the next step. to add to the weird issues reported before i am not sure if watchy is charging as it should. a red indicator led lights up when usb cable powered, however the battery symbol is always low/empty, not the expected charging symbol. the 'about watchy' info indicates 0.28 V voltage, is this expected ?

r8tt8, 2022-04-10:
@mniehus for battery and up-button, see #153.

gvenzl, 2022-06-11:
Hi all,

For what it's worth, I have exactly the same issue on my Mac OS X (Monterey).

Flashing the board via being directly attached to the USB on my laptop (USB-C with a USB-A adapter attached that has the USB-A to Micro-USB cable plugged into the board) only works very sporadically. I'd say one out of felt 50-75 times.
Every time, however, I can see the board is being reset (the screen flicking white and black just like when the upload was successful).

However, the moment I plug it into a Raspberry Pi, I can erase the flash, plug it back into my Mac and upload a new Sketch with no issues.

Would be really, really nice if that could be somehow addressed via software. I wasted about 5 hours trying all sorts of settings, different versions of the libraries, etc. and was almost about to call the board broken, before I gave the Raspberry Pi approach one last chance.

sadra-github, 2022-07-15:
Sketch uses 262434 bytes (20%) of program storage space. Maximum is 1310720 bytes.
esptool.py v3.0-dev
Serial port COM5
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
Failed uploading: uploading error: exit status 2

ktbyte, 2022-08-18:
Also having this issue, reproduced on both windows and linux, with both usb 3.0 directly and through a usb 2.0 hub. esptool erase_flash also not working on multiple computers. No raspberry PI to test with unfortunately.

nbdy, 2022-08-25:
Having the same issue with PopOS 22.04.
It works with a RPi 4 (DietPi) though. (Same USB cable)

dmesg output on RPi:
```
[153532.339464] usb 1-1.2: new full-speed USB device number 3 using xhci_hcd
[153532.447062] usb 1-1.2: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[153532.447088] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[153532.447105] usb 1-1.2: Product: CP2102N USB to UART Bridge Controller
[153532.447119] usb 1-1.2: Manufacturer: Silicon Labs
[153532.447133] usb 1-1.2: SerialNumber: 204174cd9fcaeb11853f89561d69213e
[153532.481067] usbcore: registered new interface driver usbserial_generic
[153532.481118] usbserial: USB Serial support registered for generic
[153532.488450] usbcore: registered new interface driver cp210x
[153532.488510] usbserial: USB Serial support registered for cp210x
[153532.488617] cp210x 1-1.2:1.0: cp210x converter detected
[153532.494840] usb 1-1.2: cp210x converter now attached to ttyUSB0
```

esptool output on RPi:
```
~# esptool read_mac
esptool.py v2.8
Found 2 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 4c:75:25:a7:68:b4
Enabling default SPI flash mode...
MAC: 4c:75:25:a7:68:b4
Hard resetting via RTS pin..
```

dmesg output on my PC:
```
[34797.389464] usb 1-1: new full-speed USB device number 24 using xhci_hcd
[34797.561841] usb 1-1: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[34797.561844] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[34797.561846] usb 1-1: Product: CP2102N USB to UART Bridge Controller
[34797.561847] usb 1-1: Manufacturer: Silicon Labs
[34797.561848] usb 1-1: SerialNumber: 204174cd9fcaeb11853f89561d69213e
[34797.563925] cp210x 1-1:1.0: cp210x converter detected
[34797.571997] usb 1-1: cp210x converter now attached to ttyUSB0
```

Seems like the only difference is that the device gets registered by usbcore and usbserial on the RPi?

nbdy, 2022-08-26:
Solution for me was to erase the flash with esptool on the RPi.

After uploading a watchface on my PC I'm no longer able to upload other watchfaces unless I apply the above mentioned solution.

themoonisacheese, 2022-10-08:
Adding my report to the pile:
i cannot get esptool to work at all on my desktop PC, even with an old cable, the USB hub integrated to my monitor, a 2$ usb hub that i stopped using because it was bad, usb 2 or 3, case or back ports.
it worked instantly when i used my partner's laptop on a usb 2 port.

the bad usb hub is underpowered ( it has LEDs on it that sometimes fail to turn on because of a lack of power) so i don't think this is a power issue as much as it is some sort of timing issue.

chunkysteveo, 2022-12-29:
Same issue connecting to Watchy using a Windows 11 PC ....

Ended up eventually connecting and uploading a sketch from Arduino after using a USB hub that's part of a monitor that connects via a USB SS (3) cable, but only worked when the cable was plugged into a USB 2 connection on the motherboard - all other setups of ports, USB3, USB3, Front, back etc all failed and only just the combo of a "usb hub" (i.e. the monitor has one) connected via a USB 2 port in the back worked(??!)

New watch, so set to V2 in the board settings, and the battery icon worked, voltage correct, local weather all working - all A-OK!

To help others - it's probably NOT your watch that's broken, it's probably the connection between the PC and the ESP32 - diagnose, troubleshoot, ask for help - don't just think it's broken 👍

MiraslauKavaliou, 2023-01-11:
I am having the same exact problem on my M2 Macbook air, I have to boot into Asahi Linux erase the flash with `python esptool.py erase_flash` and then boot back into macOS. And this works for only one upload in the Arduino IDE... Is there any software fix for this? or any Watchy developers that can fix this? 

Am finding this really annoying :/

lazypingu, 2023-02-24:
It works on my desktop machine with the new [esptool version 4.5](https://github.com/espressif/esptool/releases).
Erasing with `esptool.py erase flash` works fine. After that I was able to build and upload.

As a temporary workaround, since the current version of the [esp32 core](https://github.com/espressif/arduino-esp32/releases/tag/2.0.6) is bundled with 4.2.1, I just replaced (maybe make a backup before) the bundled esptool under `~/.arduino15/packages/esp32/tools/esptool_py/4.2.1` with the current version from a venv environment.

After that I was able to build and upload with arduino-cli (without a previous erase with esptool), e.g.:
```
arduino-cli compile --clean --board-options Revision=v20,UploadSpeed=115200,EraseFlash=all -b esp32:esp32:watchy ~/Arduino/libraries/Watchy/examples/WatchFaces/7_SEG
arduino-cli upload -b esp32:esp32:watchy -p /dev/ttyUSB0 ~/Arduino/libraries/Watchy/examples/WatchFaces/7_SEG
```

denics, 2024-08-13:
which version of esp32 are you running? downgrading to 2.0.17 (as suggested in the doc) did work for me without any workaround

lazypingu, 2024-08-16:
The esp core version was probably `2.0.6` back then.

I've tested it again with watchy library version `1.4.14`, esp core `2.0.17` and Watchy `V2`.
This esp core version is now bundled with esptool `4.5.1` (which needs pyserial btw).

**I can confirm this works fine for me without any workarounds.**

I've used the following parameters on arduino-cli:

```shell
arduino-cli compile --board-options Revision=v20 -b esp32:esp32:watchy <code>
arduino-cli upload -b esp32:esp32:watchy -p /dev/ttyUSB0 <code>
```

denics, 2025-07-24:
> Oh man, I wish I've read these comments 2 days ago... Is this project dead? No one could at least write a sentence in the getting started doc about being it close to impossible to flash this thing?
> 
> > You have to jump 8 times on your left foot while singing an old Chneese lullaby, lick the USB cable in the middle and then you have 5% chance that it will work.
> 
> Really? The Bluetooth OTA doesn't work and the Wifi OTA doesn't work either. This is so much waste of time and money.
> 
> BTW I love wasting my life on ESP32 projects. But this is too far.

you can have a look at this one : https://github.com/multipolygon/esphome-for-watchy I love how it works with home assistant!