sqfmi/Watchy #148: Can't build examples with Watchy library 1.3.x

rldleblanc, 2022-03-17:
I can't build 7Seg with my config changes and I thought my git was out of sync, but even trying to build the examples from the installed library fails.
```
WARNING: Category '' in library ESP RainMaker is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library WiFiProv is not valid. Setting to 'Uncategorized'
WARNING: library Rtc_Pcf8563 claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (esp32) architecture(s).
/home/rleblanc/Arduino/libraries/Watchy/src/WatchyRTC.cpp: In member function 'void WatchyRTC::clearAlarm()':
/home/rleblanc/Arduino/libraries/Watchy/src/WatchyRTC.cpp:33:22: error: 'ALARM_2' was not declared in this scope
         rtc_ds.alarm(ALARM_2);
                      ^~~~~~~
/home/rleblanc/Arduino/libraries/Watchy/src/WatchyRTC.cpp: In member function 'void WatchyRTC::_DSConfig(String)':
/home/rleblanc/Arduino/libraries/Watchy/src/WatchyRTC.cpp:93:23: error: 'SQWAVE_NONE' was not declared in this scope
     rtc_ds.squareWave(SQWAVE_NONE); //disable square wave output
                       ^~~~~~~~~~~
/home/rleblanc/Arduino/libraries/Watchy/src/WatchyRTC.cpp:93:23: note: suggested alternative: 'SIGEV_NONE'
     rtc_ds.squareWave(SQWAVE_NONE); //disable square wave output
                       ^~~~~~~~~~~
                       SIGEV_NONE
/home/rleblanc/Arduino/libraries/Watchy/src/WatchyRTC.cpp:94:21: error: 'ALM2_EVERY_MINUTE' was not declared in this scope
     rtc_ds.setAlarm(ALM2_EVERY_MINUTE, 0, 0, 0, 0); //alarm wakes up Watchy every minute
                     ^~~~~~~~~~~~~~~~~
/home/rleblanc/Arduino/libraries/Watchy/src/WatchyRTC.cpp:95:27: error: 'ALARM_2' was not declared in this scope
     rtc_ds.alarmInterrupt(ALARM_2, true); //enable alarm interrupt
                           ^~~~~~~
exit status 1
Error compiling for board ESP32 Dev Module.
```

TheKigen, 2022-03-21:
Its related to https://github.com/JChristensen/DS3232RTC/releases/tag/2.0.0
You can downgrade that library to 1.3.0 to fix it.

rldleblanc, 2022-03-21:
I was able to get it to build with the PR that I created as well.

rldleblanc, 2022-03-25:
This is fixed by PR #150

closed 2022-03-25 by rldleblanc