sqfmi/Watchy #151: 7_SEG - Battery charge indicator stays empty

8tt8r, 2022-03-25:
7_SEG watchface.
Even after 3 hours of charging, the battery indicator remains empty (as if the battery was completely flat).
It has been 3 days of operation without recharge, and it's still going.

Why doesn't battery reading work?

"About Watchy" always reports:
```
Lib Ver: v1.1.3
RTC: PCF8563
Batt: 0.28V
```

rastagraffix, 2022-03-25:
Are you sure the USB connector is fully inserted?  If you peer
into the gaps around the connector housing you should be able
to see a tiny red LED light up when the battery is charging.

I'm not sure if there is a way to detect in software if the
battery is charging.  If there is we could change the battery
indicator to show when charging is active.

But based on the voltage measurements you're reporting, it
looks like the USB connector isn't connecting properly, as
you would definitely see the voltage increase as the battery
charges.

--lyndon

smolting, 2022-03-26:
I just received my Watchy from mouser, and I have this exact same problem. Battery meter displays empty, but the watch behaves as if it's fully charged -- the red charging light switches off when it's full, all other features of the 7-seg watch face work, and it's been running fine for hours.

8tt8r, 2022-03-26:
A simple analogRead of pin 35 simply reurns a solid constant 0.

```
void setup() {
  Serial.begin(115200);
}

void loop() {
  int batt = analogRead(35);
  Serial.println(batt);
  delay(10);
}
```

It should be pin 35 because I have watchy v1.5 with the PCF8563 rtc. This is reported in the `config.h` file (watchy lib v1.3.3, NOT v1.4.0) under "V10_ADC_PIN" and "V15_ADC_PIN" pins (rows 7, 8)
[image]
.

r8tt8, 2022-03-26:
Found the problem: #150 (fix battery and up button pins).
Actual pin of the battery is 34.
I edited the `config.h` file and now it works. I think future watchy library releases shoud solve the problem.

closed 2022-03-26 by r8tt8

KidElectric, 2022-04-15:
For clarity might be worth mentioning explicitly that if you're having this problem, you also need to change the UP_BTN_PIN in config.h to be pin 35.