sqfmi/Watchy #152: Failure following "Getting Started" guide for PlatformIO

stuartpb, 2022-03-30:
When following the steps at https://watchy.sqfmi.com/docs/getting-started/#simple-watchface-example I get this output:

```log
[stuart@stugear 7seg-custom]$ pio run -t upload
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
----------------------------------------------------------------------------------------------------
Tool Manager: Installing platformio/tool-mkspiffs @ ~2.230.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-mkspiffs @ 2.230.0 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.5.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
Converting 7_SEG.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep+, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Watchy> 1.3.3
|   |-- <ESP32 BLE Arduino> 1.0.1
|   |-- <WiFiManager> 0.16.0
|   |   |-- <DNSServer> 1.1.0
|   |   |   |-- <WiFi> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <WiFi> 1.0
|   |-- <WiFi> 1.0
|   |-- <HTTPClient> 1.2
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <NTPClient> 3.1.0
|   |-- <Arduino_JSON> 0.1.0
|   |-- <GxEPD2> 1.4.5
|   |   |-- <SPI> 1.0
|   |   |-- <Adafruit GFX Library> 1.10.14
|   |   |   |-- <Adafruit BusIO> 1.11.3
|   |   |   |   |-- <Wire> 1.0.1
|   |   |   |   |-- <SPI> 1.0
|   |   |   |-- <Wire> 1.0.1
|   |   |   |-- <SPI> 1.0
|   |-- <Adafruit GFX Library> 1.10.14
|   |   |-- <Adafruit BusIO> 1.11.3
|   |   |   |-- <Wire> 1.0.1
|   |   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|   |-- <DS3232RTC> 0.0.0-alpha+sha.261ca7d0e6
|   |   |-- <Time> 1.6.1
|   |   |-- <Wire> 1.0.1
|   |-- <Time> 1.6.1
|   |-- <Rtc_Pcf8563> 0.0.0-alpha+sha.6b5adc2ced
|   |   |-- <Wire> 1.0.1
|-- <WifiManager> 2.0.3-alpha+sha.9af7253
|   |-- <WiFi> 1.0
|   |-- <WebServer> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <FS> 1.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <WiFi> 1.0
|-- <WiFiManager> 0.16.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <WiFi> 1.0
|   |-- <WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <WiFi> 1.0
|-- <HTTPClient> 1.2
|   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <WiFiClientSecure> 1.0
|   |-- <WiFi> 1.0
|-- <NTPClient> 3.1.0
|-- <Arduino_JSON> 0.1.0
|-- <GxEPD2> 1.4.5
|   |-- <SPI> 1.0
|   |-- <Adafruit GFX Library> 1.10.14
|   |   |-- <Adafruit BusIO> 1.11.3
|   |   |   |-- <Wire> 1.0.1
|   |   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|-- <Adafruit GFX Library> 1.10.14
|   |-- <Adafruit BusIO> 1.11.3
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <SPI> 1.0
|-- <Wire> 1.0.1
|-- <DS3232RTC> 0.0.0-alpha+sha.261ca7d0e6
|   |-- <Time> 1.6.1
|   |-- <Wire> 1.0.1
|-- <Time> 1.6.1
|-- <Rtc_Pcf8563> 0.0.0-alpha+sha.6b5adc2ced
|   |-- <Wire> 1.0.1
|-- <ESP32 BLE Arduino> 1.0.1
Building in release mode
Compiling .pio/build/esp32dev/src/7_SEG.ino.cpp.o
Compiling .pio/build/esp32dev/src/Watchy_7_SEG.cpp.o
Generating partitions .pio/build/esp32dev/partitions.bin
Compiling .pio/build/esp32dev/lib79d/BLE/BLE2902.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLE2904.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLEAddress.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLEAdvertisedDevice.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLEAdvertising.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLEBeacon.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLECharacteristic.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLECharacteristicMap.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLEClient.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLEDescriptor.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLEDescriptorMap.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLEDevice.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLEEddystoneTLM.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLEEddystoneURL.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLEExceptions.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLEHIDDevice.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLERemoteCharacteristic.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLERemoteDescriptor.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLERemoteService.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLEScan.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLESecurity.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLEServer.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLEService.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLEServiceMap.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLEUUID.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLEUtils.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/BLEValue.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/FreeRTOS.cpp.o
Compiling .pio/build/esp32dev/lib79d/BLE/GeneralUtils.cpp.o
Compiling .pio/build/esp32dev/lib0db/WiFi/ETH.cpp.o
Compiling .pio/build/esp32dev/lib0db/WiFi/WiFi.cpp.o
Compiling .pio/build/esp32dev/lib0db/WiFi/WiFiAP.cpp.o
Compiling .pio/build/esp32dev/lib0db/WiFi/WiFiClient.cpp.o
Compiling .pio/build/esp32dev/lib0db/WiFi/WiFiGeneric.cpp.o
Compiling .pio/build/esp32dev/lib0db/WiFi/WiFiMulti.cpp.o
Compiling .pio/build/esp32dev/lib0db/WiFi/WiFiSTA.cpp.o
Compiling .pio/build/esp32dev/lib0db/WiFi/WiFiScan.cpp.o
Compiling .pio/build/esp32dev/lib0db/WiFi/WiFiServer.cpp.o
Compiling .pio/build/esp32dev/lib0db/WiFi/WiFiUdp.cpp.o
Archiving .pio/build/esp32dev/lib79d/libBLE.a
Indexing .pio/build/esp32dev/lib79d/libBLE.a
Compiling .pio/build/esp32dev/lib08d/DNSServer/DNSServer.cpp.o
Compiling .pio/build/esp32dev/lib64a/WiFiManager/WiFiManager.cpp.o
In file included from .pio/libdeps/esp32dev/WiFiManager/WiFiManager.cpp:13:0:
.pio/libdeps/esp32dev/WiFiManager/WiFiManager.h:16:25: fatal error: ESP8266WiFi.h: No such file or directory

*********************************************************************
* Looking for ESP8266WiFi.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP8266WiFi.h"
* Web  > https://registry.platformio.org/search?q=header:ESP8266WiFi.h
*
*********************************************************************

compilation terminated.
Compiling .pio/build/esp32dev/liba52/WiFiClientSecure/WiFiClientSecure.cpp.o
*** [.pio/build/esp32dev/lib64a/WiFiManager/WiFiManager.cpp.o] Error 1
==================================== [FAILED] Took 9.07 seconds ====================================
```

stuartpb, 2022-03-30:
I fixed this by pinning `https://github.com/tzapu/WiFiManager.git#v2.0.10-beta` instead of `2.0.3-alpha` in `platformio.ini`.

(I also had to run `sudo usermod -aG uucp $USER` as described at https://docs.platformio.org/en/latest/faq.html#platformio-udev-rules and reboot, but that's a local-system-specific problem/solution that the PIO CLI prompts you to solve.)

mrmowgli, 2022-05-13:
Thumbs up on this issue, I burned a lot of time on the WIFIManager version which caused compile errors. Also, now I have it building it isn't actually completing an upload, it varies from 59-74% before failing.

Using this as my Platformio.ini:
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps =
    sqfmi/Watchy @ 1.4.1 ; Pinned version to ensure we don't pull broken code
    https://github.com/tzapu/WiFiManager.git#v2.0.11-beta ; Pinned for the same reason
lib_ldf_mode = deep+
board_build.partitions = min_spiffs.csv
upload_speed = 3000000
upload_port =  /dev/ttyUSB0
monitor_port = /dev/ttyUSB0
monitor_speed = 115200
monitor_filters = esp32_exception_decoder 
```

And the upload results: 
```
pio run -t upload
Processing esp32dev (platform: espressif32; board: esp32dev; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (4.2.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20002.220503 (2.0.2) 
 - tool-esptoolpy @ 1.30300.0 (3.3.0) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
Converting 7_SEG.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep+, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Watchy @ 1.4.1
|   |-- ESP32 BLE Arduino @ 2.0.0
|   |-- WiFiManager @ 2.0.11-beta+sha.13b4307
|   |   |-- WiFi @ 2.0.0
|   |   |-- Update @ 2.0.0
|   |   |-- WebServer @ 2.0.0
|   |   |   |-- WiFi @ 2.0.0
|   |   |   |-- FS @ 2.0.0
|   |   |-- DNSServer @ 2.0.0
|   |   |   |-- WiFi @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- Update @ 2.0.0
|   |-- WebServer @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- DNSServer @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- HTTPClient @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |   |-- WiFiClientSecure @ 2.0.0
|   |   |   |-- WiFi @ 2.0.0
|   |-- WiFiClientSecure @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- NTPClient @ 3.2.1
|   |-- Arduino_JSON @ 0.1.0
|   |-- GxEPD2 @ 1.4.5
|   |   |-- SPI @ 2.0.0
[... 144 lines cut ...]
Writing at 0x000d3146... (43 %)
Writing at 0x000d87a0... (45 %)
Writing at 0x000de62a... (46 %)
Writing at 0x000e3a30... (48 %)
Writing at 0x000e90a2... (50 %)
Writing at 0x000eec95... (51 %)
Writing at 0x000f4613... (53 %)
Writing at 0x000fa44d... (54 %)
Writing at 0x00101b76... (56 %)
Writing at 0x00107b3b... (57 %)
Writing at 0x0010d91a... (59 %)
Writing at 0x0011378e... (60 %)
Writing at 0x00118e04... (62 %)
Writing at 0x0011e2d6... (63 %)
Writing at 0x001237af... (65 %)
Writing at 0x00129626... (66 %)
Writing at 0x0012f139... (68 %)
Writing at 0x00134871... (69 %)
Writing at 0x0013a689... (71 %)
Writing at 0x0013ffd0... (72 %)
Writing at 0x0014561a... (74 %)

A fatal error occurred: Serial data stream stopped: Possible serial noise or corruption.
*** [upload] Error 2
```

mrmowgli, 2022-05-13:
Here's the final version of platformio.ini that worked for me:
```
[env:esp32dev]
platform = espressif32
board = pico32
; board = esp32dev
framework = arduino
lib_deps =
    sqfmi/Watchy @ 1.4.1 ; Pinned version to ensure we don't pull broken code
    https://github.com/tzapu/WiFiManager.git#v2.0.11-beta ; Pinned for the same reason
lib_ldf_mode = deep+
board_build.partitions = min_spiffs.csv

; upload_speed = 3000000
upload_port =  /dev/ttyUSB0
monitor_port = /dev/ttyUSB0
; monitor_speed = 115200
monitor_speed = 921600
monitor_filters = esp32_exception_decoder

board_build.f_cpu = 240000000L
board_build.f_flash = 80000000L
build_flags = -DCORE_DEBUG_LEVEL=0
```

GuruSR, 2022-07-24:
platform_packages = framework-arduinoespressif32 @ 3.20002.220503

Might want to include that in your platformio.ini as well.

Also, if your Watchy is boot loop crashing, upload only at 115200 otherwise corruption or failure will happen, because the crash watchdog code sets the ESP32's UART to 115200 to send the error out to the UART, if you're uploading at another baud rate, it'll get lost.

That locks the ESP32 at 2.0.2.

muffins, 2023-11-10:
> Here's the final version of platformio.ini that worked for me:
>
> ```
> [env:esp32dev]
> platform = espressif32
> board = pico32
> ; board = esp32dev
> framework = arduino
> lib_deps =
>     sqfmi/Watchy @ 1.4.1 ; [...]

This worked for me! Thank you very much for posting!! I'll drop my own config, but I've been pulling my hair out for months trying to get my Watchy firmware uploaded, which I'd never been able to get much past like 40 - 70% uploaded. I was finally able to at least get a basic face up:

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps =
    sqfmi/Watchy @ 1.4.7 ; 1.4.7+ Fixes the WIDTH_VISIBLE bug
    https://github.com/tzapu/WiFiManager.git#v2.0.11-beta ; Pinned to not pull broken code
lib_ldf_mode = deep+
board_build.partitions = min_spiffs.csv

upload_port =  /dev/tty.usbserial-1420
monitor_port = /dev/tty.usbserial-1420
monitor_speed = 921600
monitor_filters = esp32_exception_decoder

board_build.f_cpu = 240000000L
board_build.f_flash = 80000000L
build_flags = -DCORE_DEBUG_LEVEL=0
```