sqfmi/Watchy #153: Battery does not appear to be charging

stuartpb, 2022-03-31:
I left the Watchy plugged in overnight, but going to "About Watchy" still shows "Batt: 0.28V". DO I need to do a full discharge / charge cycle to calibrate or something like that?

rastagraffix, 2022-03-31:
Stuart P. Bentley writes:

> I left the Watchy plugged in overnight, but going to "About Watchy" still sho
> ws "Batt: 0.28V". DO I need to do a full discharge / charge cycle to calibrat
> e or something like that?

I would go for the full discharge/re-charge cycle.  If it's not working
after that you might want to consider that you just have a bad battery ...

--lyndon

electronjoe, 2022-04-01:
I experienced this issue -and found that the wrong ADC pin was being used for my Watchy build.

This is a dupe issue of #151.

I am not using Arduino IDE - but my solution was to set the correct precompiler define (`ARDUINO_WATCHY_V20`) - as this sets pins correctly for the most recent watch builds.

stuartpb, 2022-04-02:
While I am seeing issues that suggest this pin mapping issue is the source, the 7-Seg watchface doesn't seem to go through the [Espressif ESP32 Arduino Core](https://github.com/espressif/arduino-esp32) that uses that variable - only the Watchy module from this repo, which [still sniffs out pin layouts based on the RTC model](https://github.com/sqfmi/Watchy/blob/master/src/Watchy.cpp#L613-619).

synthead, 2022-05-07:
If I were to vote on priority of bug fixes, this should be far up.  This is a breaking change in the UX.

I just received a replacement Watchy from Mouser because I assumed there was a hardware defect with my first one, and the replacement has the same behavior.  The instructions say to ensure the watch has a full charge before using it, so if users are following the directions verbatim, then every Watchy being shipped could be considered unusable.

I would have never guessed that something as important as the battery meter would be broken on delivery.  I honestly don't know why this wasn't tested and/or considered a blocker to shipping the units.

gvenzl, 2022-06-12:
I also just got my new Watchy and am running into the same issue.
However, it doesn't seem to be related to the wrong pin in my case.
Regardless of whether I read from PIN 33, 34, or 35, the voltage is reported to be just about 0.30V, while I checked the battery with a Multimeter and it has 4.1V in it.

I added the following code to the setup(), after checking how Watchy is retrieving the battery voltage:

```
#include "Watchy_7_SEG.h"
#include "settings.h"

Watchy7SEG watchy(settings);

void setup(){

  Serial.begin(115200);

  Serial.println("Battery pin:");
  Serial.println(BATT_ADC_PIN, DEC);
  
  float voltage33 = analogReadMilliVolts(33);
  Serial.println("Analoge voltage 33: ");
  Serial.println(voltage33, DEC);

  float voltage34 = analogReadMilliVolts(34);
  Serial.println("Analoge voltage 34: ");
  Serial.println(voltage34, DEC);

  float voltage35 = analogReadMilliVolts(35);
  Serial.println("Analoge voltage 35: ");
  Serial.println(voltage35, DEC);

  float vol = watchy.getBatteryVoltage();
  Serial.println("Watchy voltage: ");
  Serial.println(vol, DEC);

  watchy.init();

}

void loop() {}
```

Does anybody have an idea why none of the pins report the correct voltage?

I used the same methods to retrieve them as the Watchy code does, i.e. at the bottom you will see the `watchy.getBatteryVoltage()` that is called by `Watchy7SEG::drawBattery()` while the rest uses the `analogReadMilliVolts()` that is called from `Watchy::getBatteryVoltage`.

The result is wrong for any pin of the 3 versions of Watchy :/

```
14:52:57.526 -> Battery pin:
14:52:57.526 -> 34
14:52:57.526 -> Analoge voltage 33: 
14:52:57.526 -> 639.0000000000
14:52:57.526 -> Analoge voltage 34: 
14:52:57.526 -> 234.0000000000
14:52:57.526 -> Analoge voltage 35: 
14:52:57.526 -> 142.0000000000
14:52:57.526 -> Watchy voltage: 
14:52:57.526 -> 0.3919999897
14:53:40.090 -> Battery pin:
14:53:40.090 -> 34
14:53:40.090 -> Analoge voltage 33: 
14:53:40.090 -> 627.0000000000
14:53:40.090 -> Analoge voltage 34: 
14:53:40.128 -> 187.0000000000
14:53:40.128 -> Analoge voltage 35: 
14:53:40.128 -> 142.0000000000
14:53:40.128 -> Watchy voltage: 
14:53:40.128 -> 0.3420000076
```

tlhintoq, 2022-07-30:
29jul2022 - Brandy new watchy right out of the box.
Same behavior - no customization done yet.
UI shows battery depleted.
About shows battery 0.28v

Earlier comments say its a pin issue that can be confirmed by the top-right switch not working as expected.  As I can work the main menu both up and down (sw3, sw4) all indications are the pin definitions would be correct.

@gvenzl 's comment would indicate the battery does charge - just that we don't know what the actual level is.  Joy.

gvenzl, 2022-08-20:
Yeah, the battery on my Watchy definitely does charge, I measured it with a multimeter and it does have and hold the charge for a couple of days. Still, no indication of how to retrieve the battery voltage. Anybody any ideas?

gvenzl, 2022-09-10:
Hey @sqfmi,

It seems like several users observe the same issue with not being able to read the battery charge. Is this something you could check out whether you can reproduce it at your end maybe?

shtrom, 2022-10-19:
> 29jul2022 - Brandy new watchy right out of the box. Same behavior - no customization done yet. UI shows battery depleted. About shows battery 0.28v
> 
> Earlier comments say its a pin issue that can be confirmed by the top-right switch not working as expected.

I observed the same behaviour.

The key to the solution was in here:

> While I am seeing issues that suggest this pin mapping issue is the source, the 7-Seg watchface doesn't seem to go through the [Espressif ESP32 Arduino Core](https://github.com/espressif/arduino-esp32) that uses that `ARDUINO_WATCHY_V20` variable - only the Watchy module from this repo, which [still sniffs out pin layouts based on the RTC model](https://github.com/sqfmi/Watchy/blob/master/src/Watchy.cpp#L613-619).

I had `ARDUINO_WATCHY_V20` set after upgrading the ESP32 module in Arduino. I set it back to `ARDUINO_WATCHY_V10`, and all got fixed! I guess the same issue can happen on older versions (mid-2021 for me), too. The key is a mismatch between the hardware and the compilation target.

[image]

Paul1th0, 2022-11-10:
Hi there. I just got the watchy yesterday and had the same issue with the battery status not showing correctly. For my it was the other way around. The Board Revision must have been done with Watchy v1.0 (upper right button was also not working right) after I loaded the 7_SEG on the watchy again with Watchy v2.0 it both worked.

sqfmi, 2022-12-29:
@gvenzl have you tried the latest release and selecting the Watchy board revision, if the issue still persists? If your board was bought recently, it should be v2.0.

In your test code the results do look off, can you try using ```analogRead(33)``` or ```analogRead(34)``` with the battery plugged in and see what the results are? The correct value should be ((4.2V / 2) / 3.3V)*4095 ~= 2605. If possible, please share with us a pic of the PCB to spot any potential hardware issues. Thanks!

gvenzl, 2023-01-17:
Hey @sqfmi,

Yes, it works now. I just uploaded `1.4.6` and the battery bars are now showing on my watch. I've used revision `v2.0` as you said.

Just for reference, I've also modified the program to include the `analogRead()`:

```cpp
#include "Watchy_7_SEG.h"
#include "settings.h"

Watchy7SEG watchy(settings);

void setup(){
  Serial.begin(115200);

  Serial.println("Battery pin:");
  Serial.println(BATT_ADC_PIN, DEC);
  
  float voltage33 = analogReadMilliVolts(33);
  float analog33 = analogRead(33);
  Serial.println("Analoge voltage 33: ");
  Serial.println(voltage33, DEC);
  Serial.println("Analog read 33: ");
  Serial.println(analog33, DEC);

  float voltage34 = analogReadMilliVolts(34);
  float analog34 = analogRead(34);
  Serial.println("Analoge voltage 34: ");
  Serial.println(voltage34, DEC);
  Serial.println("Analog read 34: ");
  Serial.println(analog34, DEC);

  float voltage35 = analogReadMilliVolts(35);
  float analog35 = analogRead(35);
  Serial.println("Analoge voltage 35: ");
  Serial.println(voltage35, DEC);
  Serial.println("Analog read 35: ");
  Serial.println(analog35, DEC);

  float vol = watchy.getBatteryVoltage();
  Serial.println("Watchy voltage: ");
  Serial.println(vol, DEC);

  watchy.init();

}

void loop(){}
```

Here the output:

```
Battery pin:
34
Analoge voltage 33: 
864.0000000000
Analog read 33: 
1057.0000000000
Analoge voltage 34: 
2094.0000000000
Analog read 34: 
2441.0000000000
Analoge voltage 35: 
142.0000000000
Analog read 35: 
0.0000000000
Watchy voltage: 
4.1880002022
Battery pin:
34
Analoge voltage 33: 
816.0000000000
Analog read 33: 
1041.0000000000
Analoge voltage 34: 
2098.0000000000
Analog read 34: 
2442.0000000000
Analoge voltage 35: 
142.0000000000
Analog read 35: 
0.0000000000
Watchy voltage: 
4.1960000992
```

sqfmi, 2023-01-17:
Thanks @gvenzl! Glad to know it's been resolved. We'll update the docs and FAQ

onionhammer, 2023-06-14:
Is there a link to updated docs on this? I just got a new watchy from mouser today, and it has this issue as well

ckirtley, 2024-04-24:
Mine was charging fine (and showing a blue flashing light), but suddenly it's not doing either. I haven't changed anything recentlym and it was charging fine before. I wondered if it was hardware issue - maybe the USB socket?