sqfmi/Watchy #157: Compiling error: 'ALARM_2' was not declared

KidElectric, 2022-04-14:
Hello I am trying to compile and upload a new watchface but get the following issue coming up no matter which one I try. Does anyone know why this might be happening? Thank you!

Arduino: 1.8.15 (Linux), Board: "ESP32 Dev Module, Disabled, Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None"

WARNING: library Rtc_Pcf8563 claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
/home/brian/snap/arduino/current/Arduino/libraries/Watchy/src/WatchyRTC.cpp: In member function 'void WatchyRTC::clearAlarm()':
/home/brian/snap/arduino/current/Arduino/libraries/Watchy/src/WatchyRTC.cpp:33:22: error: 'ALARM_2' was not declared in this scope
         rtc_ds.alarm(ALARM_2);
                      ^~~~~~~
/home/brian/snap/arduino/current/Arduino/libraries/Watchy/src/WatchyRTC.cpp: In member function 'void WatchyRTC::_DSConfig(String)':
/home/brian/snap/arduino/current/Arduino/libraries/Watchy/src/WatchyRTC.cpp:93:23: error: 'SQWAVE_NONE' was not declared in this scope
     rtc_ds.squareWave(SQWAVE_NONE); //disable square wave output
                       ^~~~~~~~~~~
/home/brian/snap/arduino/current/Arduino/libraries/Watchy/src/WatchyRTC.cpp:93:23: note: suggested alternative: 'SIGEV_NONE'
     rtc_ds.squareWave(SQWAVE_NONE); //disable square wave output
                       ^~~~~~~~~~~
                       SIGEV_NONE
/home/brian/snap/arduino/current/Arduino/libraries/Watchy/src/WatchyRTC.cpp:94:21: error: 'ALM2_EVERY_MINUTE' was not declared in this scope
     rtc_ds.setAlarm(ALM2_EVERY_MINUTE, 0, 0, 0, 0); //alarm wakes up Watchy every minute
                     ^~~~~~~~~~~~~~~~~
/home/brian/snap/arduino/current/Arduino/libraries/Watchy/src/WatchyRTC.cpp:95:27: error: 'ALARM_2' was not declared in this scope
     rtc_ds.alarmInterrupt(ALARM_2, true); //enable alarm interrupt
                           ^~~~~~~
Multiple libraries were found for "WiFi.h"
 Used: /home/brian/snap/arduino/61/.arduino15/packages/esp32/hardware/esp32/2.0.2/libraries/WiFi
 Not used: /snap/arduino/61/libraries/WiFi
exit status 1
Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

gpsamu, 2022-04-15:
Just received my Watchy also and I'm having the exact same issue (same error).  I've setup a fresh Arduino install (1.8.19) and have installed the ESP 32 (2.0.2) an Watchy (1.3.3) libraries.  Any help would be greatly appreciated as I wanted to try some other watch faces.

Swepedia, 2022-04-15:
So if you look at #148 they talk about downgrading DS3232RTC. I downgraded to 1.3.0 and it worked.

gpsamu, 2022-04-15:
> So if you look at #148 they talk about downgrading DS3232RTC. I downgraded to 1.3.0 and it worked.

That appears to have done the trick! Many thanks.

KidElectric, 2022-04-15:
Awesome I will give that a shot as well thank you!!

closed 2022-04-15 by KidElectric