sqfmi/Watchy #158: Watchy fully charged but battery icon shows no bars

KidElectric, 2022-04-14:
Title says it all. Watchy has been running for 2-3 days after full charge but the battery icon never changed to show bars or changed to show bars going down. It has remained an empty battery icon since I opened the box. I've heard people talk about the battery calibration needing fine-tuning ... is this what is happening here? It seems like this would be difficult to adjust manually as I would need to divide by a very small number in there somewhere to get a value of 0 to turn into some bars.

Swepedia, 2022-04-15:
I believe this is a dupe. If you go look at #153 and #151 they have info on fixing this. The PR #150 fixes the issue, so if you pull that code you should be able to get it working. I was also having this issue and the PR seemed to fix it. You'll also have to install the ArduinoJson library, which is different from the Arduino_JSON library. If you're using Arduino IDE just search for it to install.

KidElectric, 2022-04-15:
Amazing, thank you I didn't see those before. It sounds identical (0.28V shown on about screen and everything). Great to know it can be fixed with a pin change!

closed 2022-04-15 by KidElectric