sqfmi/Watchy #163: How to use 12 hour time?

synthead, 2022-05-07:
Is there a way to use 12h time with Watchy without writing something myself?

I don't see anything about this in the documentation, so I thought I'd file an issue and ask.

bushmango, 2022-05-11:
Here's some code from the 7SEG example that may help:

void Watchy7SEG::drawTime(){
    display.setFont(&DSEG7_Classic_Bold_53);
    display.setCursor(5, 53+5);
    int displayHour;
    if(HOUR_12_24==12){
      displayHour = ((currentTime.Hour+11)%12)+1;
    } else {
      displayHour = currentTime.Hour;
    }
    if(displayHour < 10){
        display.print("0");
    }
    display.print(displayHour);
    display.print(":");
    if(currentTime.Minute < 10){
        display.print("0");
    }
    display.println(currentTime.Minute);
}

synthead, 2022-05-12:
I was hoping to find some kind of way to set this without doing it myself, but it looks like Watchy doesn't have native support to do this.

It would be great to have a setting in the menu, and maybe something lie `currentTime.Is12Hour` available to the faces.  Maybe even have `currentHour.Hour12` and `currentHour.IsPM` helper functions, too.

This approach has these benefits:

- `currentTime.Hour` remains the same, so there would be no breaking changes
- The added helper functions encourages the DRY principle between watch faces
- User-configured settings will persist between watch faces if the above suggestions are implemented

KidElectric, 2022-09-04:
I agree that this should be a setting in the menu and at the very least ... a setting in the settings.h file? Changing 12/24 hour time is a nearly universal feature to digital watches. You might say right after telling the time, you decide whether hours are base 12 time or base 24.