sqfmi/Watchy #169: Unable to connect to Wifi.  Advice?

undeliverable, 2022-05-20:
I'm on about my 30th try on this.   I have tried with the stock firmware and updated to the latest samples on GitHub.

  I am able to connect to the configuration AP, however after I enter my AP information and password the connection times out.  It is a 2.4Ghz access point and have not had any issues with any other devices.  Its using WPA2 personal security, and the access point shows up in the list of access points in the configuration page.  Anything else I can try?

ernestum, 2022-06-02:
I am having the same issue.

sukhpalhub, 2022-12-27:
Are you still having this issue?

If so, try navigating to 192.168.4.1 in the browser after connecting to the Watchy AP wifi and then select Info.
It will display Access Point MAC so make sure to allow this MAC address if you are having MAC address filtering.

Then click 'Erase Wifi Config' and wait for Watchy to restart itself. Once it happen, try the wifi setup again and it should hopefully work.

nicokaniak, 2025-03-09:
But I cannot figure out what we should put on the SSId and password. Should it be the ones from our local wifi?