sqfmi/Watchy #171: Buttons stop working and screen flashes in 1s intervals after flashing examples

ernestum, 2022-06-02:
I just got my Watchy and right after flashing the first example, the buttons become unresponsive (I can't enter the menu) and the screen seems to update every second instead of every minute.
Before everything worked correctly (apart from WiFi not logging in)

tunnelpuzzle, 2022-07-06:
This might fix the screen updating every few seconds: https://github.com/sqfmi/watchy-docs/pull/42

ITCactus, 2022-07-11:
@ernestum if you flashed watchy from ArduinoIDE, you should probably change "config.h" in the arduino libraries (e.g. in "Arduino\libraries\Watchy\src\config.h") and try to play with "ARDUINO_WATCHY_Vxx" define.
for me, i had to set on line 17
`#define ARDUINO_WATCHY_V15` 
to make it work with my unit.
not sure where the "ARDUINO_WATCHY_Vxx" is supposed to be properly defined...

if this solved issue with buttons but not with the screen flickering, try to downgrade "esp32" board version in "Boards Manager" to "2.0.2". there is issue in "2.0.3" as mentioned in ["Getting Started"](https://watchy.sqfmi.com/docs/getting-started/#arduino-setup), but the issue remains also on the latest version (2.0.4 as of now).

of course, after the changes, you have to build and upload the desired watchface using ArduinoIDE to your watchy via usb.

hope this helps to solve the issue

Bertruss, 2022-07-15:
> @ernestum if you flashed watchy from ArduinoIDE, you should probably change "config.h" in the arduino libraries (e.g. in "Arduino\libraries\Watchy\src\config.h") and try to play with [...]

this is what I needed to know, that the issue persists in 2.0.4

sqfmi, 2022-10-15:
This issue should be fixed in v1.4.2 and works with the latest versions of ESP32 core. Please re-open if issues persists.

closed 2022-10-15 by sqfmi