sqfmi/Watchy #174: Menu constantly scrolling upwards.

Bertruss, 2022-07-15:
I have checked the upper right hand button for a short but could not detect one, and yet once in menu, the selection constantly cycles up.

Bertruss, 2022-07-15:
Just checked the /Arduino/libraries/Watchy/src/config.h, it seem the UP_BTN_PIN definition depends on the watchy version you have?

```
#if !defined(ARDUINO_WATCHY_V10) && !defined(ARDUINO_WATCHY_V15) && !defined(ARDUINO_WATCHY_V20)
    #warning Hardware revision is not defined at the project level. Using hard-coded value
    #define ARDUINO_WATCHY_V20
#endif

#if defined (ARDUINO_WATCHY_V10)
    #define UP_BTN_PIN 32
    #define BATT_ADC_PIN 33
    #define UP_BTN_MASK  GPIO_SEL_32
    #define RTC_TYPE 1 //DS3231
#elif defined (ARDUINO_WATCHY_V15)
    #define UP_BTN_PIN 32
    #define BATT_ADC_PIN 35
    #define UP_BTN_MASK  GPIO_SEL_32
    #define RTC_TYPE 2 //PCF8563
#elif defined (ARDUINO_WATCHY_V20)
    #define UP_BTN_PIN 35
    #define BATT_ADC_PIN 34
    #define UP_BTN_MASK  GPIO_SEL_35
    #define RTC_TYPE 2 //PCF8563
#endif
```

Bertruss, 2022-07-15:
That fixed it :D

ananthb, 2022-09-16:
I have the same issue! I thought my watch was busted.

ananthb, 2022-09-16:
@Bertruss how did you fix it?

Bertruss, 2022-09-16:
@ananthb Change the line that says ```#define ARDUINO_WATCHY_V20``` to ```#define ARDUINO_WATCHY_V15```. 

Could also be v10 but that seems unlikely at this point.

ananthb, 2022-09-16:
I #defined the V15 label in the ino file before any includes. The Hardware Revision #warning went away but my watch is still scrolling. I think mine is a V15 cos it has that particular RTC.

portothree, 2022-09-22:
Same issue here and changing it to `V10` or `V15` did not fix my problem, in fact I was not even able to compile with that.
```
.pio/libdeps/esp32dev/Watchy/src/config.h:19:6: warning: #warning Hardware revision is not defined at the project level. Using hard-coded value [-Wcpp]
     #warning Hardware revision is not defined at the project level. Using hard-coded value
      ^
Compiling .pio/build/esp32dev/libd1e/Watchy/bma.cpp.o
.pio/libdeps/esp32dev/Watchy/src/WatchyRTC.cpp: In member function 'void WatchyRTC::read(tmElements_t&)':
.pio/libdeps/esp32dev/Watchy/src/WatchyRTC.cpp:50:13: error: 'class Rtc_Pcf8563' has no member named 'getDateTime'
     rtc_pcf.getDateTime();
```

Bertruss, 2022-09-22:
@portothree So the define is firing off correctly but you're missing dependencies related to the RTC? Honestly it took me a couple hours to get PIO working, but you'll notice my fix was specifically for Arduino where SQFMI published materials have a working walkthrough for getting dependencies resolved. What does your platformio.ini look like? It looks like you may be using an incompatible version of the RTC_Pcf8563 library. 

@ananthb My limited knowledge of C makes me think that should have worked but I'd recommend going to those lines of "config.h" in the location /Arduino/libraries/Watchy/src/config.h and just commenting out that conditional and placing the revision number ```#define``` line there. 

I'd recommend both of you check out the discord, it'll be faster and more knowledgeable than checking here... asking noob me.  I'd really have liked someone from SQFMI to have addressed this issue by modifying their guide slightly to address the opacity of people receiving NIB older hardware revisions that don't work out of the box with the published codebase. That's why I didn't close this ticket, despite having "solved it", I thought a dev might chime in. It's also possible my solution is out of date relative to recent changes to the codebase, as I published this issue in july.

the discord ---> https://discord.gg/ZXDegGV8E7

portothree, 2022-09-23:
I'll raise this on the discord server, thank you.

Yes, does not look like `1.4.1` or latest `master` version works out of the box with the hardware I have.

Here's how my `platformio.ini` looks like:
```
[env:esp32dev]
platform = espressif32 @ ~3.5.0
board = esp32dev
framework = arduino
lib_deps = 
		sqfmi/Watchy @ 1.4.1
		https://github.com/tzapu/WiFiManager.git#v2.0.11-beta
lib_ldf_mode = deep+
board_build.partitions = min_spiffs.csv
```

portothree, 2022-09-23:
Looks like platformio is doing a override for RTC library.
I tried including `https://github.com/orbitalair/Rtc_Pcf8563` in my `lib_deps` without success

sqfmi, 2022-10-08:
Updated in 1.4.2 to use ```getDate```. Watchy board version is now also available in the Arduino IDE with the latest ESP32 Arduino Core. Please re-open if issue persists, thanks!

closed 2022-10-08 by sqfmi