sqfmi/Watchy #175: [feature request] watchy 2.0 with touch and light

ITCactus, 2022-07-25:
after i dropped my watchy on the carpet... the screen started behaving odd... i assume i need a screen replacement, but there is no available as for now at [sqfmi shop](https://shop.sqfmi.com/products/watchy-screen).

however, i see that "GDEH0154D67" is a part-number screen by good-display co.
and there are addon modules compatible with that screens like [front-light](https://www.good-display.com/product/257.html) and [touch screen](https://www.good-display.com/product/258.html).

so, it would be great to see the Watchy 2.0 upgrade with those modules!
basically, the front-light (or back-light) is great in dark environment.
and touch screen may help with more user-friendly ui (e.g. with LVGL support).

shtrom, 2022-10-24:
I gotta say, backlighting is the feature I miss the most from my Pebble. I still have that occasional wrist flick muscle memory when I want to know the time in the dark.

scott8, 2023-12-10:
Same here. It really needs a backlight!

shtrom, 2024-08-23:
There is a backlight-enabled pin-for-pin compatible version for the v<=2.0 module here: GDEH0154D67-FL https://www.e-paper-display.com/GDEH0154D67%20V2.0%20Specificationc58c.pdf?method=picker&flag=all&id=47775e88-2e8c-44e8-a410-2e24006ed9a7&fileId=805&v=1.zip

csm10495, 2024-09-10:
> There is a backlight-enabled pin-for-pin compatible version for the v<=2.0 module here: GDEH0154D67-FL https://www.e-paper-display.com/GDEH0154D67%20V2.0%20Specificationc58c.pdf?method=picker&flag=all&id=47775e88-2e8c-44e8-a410-2e24006ed9a7&fileId=805&v=1.zip

Is there a way to make it light up on wrist flick though?

scott8, 2024-09-11:
> There is a backlight-enabled pin-for-pin compatible version for the v<=2.0 module here: GDEH0154D67-FL https://www.e-paper-display.com/GDEH0154D67%20V2.0%20Specificationc58c.pdf?method=picker&flag=all&id=47775e88-2e8c-44e8-a410-2e24006ed9a7&fileId=805&v=1.zip

A light seems to require a second ribbon cable, which the circuit board does not support. I see no indication that the display mentioned has any kind of light, front or back. The one I ordered does not.