sqfmi/Watchy #198: Compilation Error for 7-Seg

cromulus, 2022-12-25:
Mac os x 13.1
Arduino 2.0.3
Setup according to https://watchy.sqfmi.com/docs/getting-started/

```
$ lsusb | grep Silicon
Bus 000 Device 005: ID 10c4:ea60 Silicon Laboratories, Inc. CP2102N USB to UART Bridge Controller  Serial: 5e9ab75b45b2eb118bfb745318997a59

```

This is the error:

```
WARNING: library Rtc_Pcf8563 claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
/Users/bill/Documents/Arduino/libraries/Watchy/src/Watchy.cpp: In member function 'weatherData Watchy::getWeatherData(String, String, String, String, String, uint8_t)':
/Users/bill/Documents/Arduino/libraries/Watchy/src/Watchy.cpp:620:48: error: ambiguous overload for 'operator=' (operand types are 'String' and 'JSONVar')
             responseObject["weather"][0]["main"];
                                                ^
In file included from /Users/bill/Library/Arduino15/packages/esp32/hardware/esp32/2.0.6/cores/esp32/Arduino.h:166,
                 from /Users/bill/Documents/Arduino/libraries/Watchy/src/Watchy.h:4,
                 from /Users/bill/Documents/Arduino/libraries/Watchy/src/Watchy.cpp:1:
/Users/bill/Library/Arduino15/packages/esp32/hardware/esp32/2.0.6/cores/esp32/WString.h:102:18: note: candidate: 'String& String::operator=(const String&)'
         String & operator =(const String &rhs);
                  ^~~~~~~~
/Users/bill/Library/Arduino15/packages/esp32/hardware/esp32/2.0.6/cores/esp32/WString.h:103:18: note: candidate: 'String& String::operator=(const char*)'
         String & operator =(const char *cstr);
                  ^~~~~~~~

exit status 1

Compilation error: exit status 1
```

Any ideas?

cromulus, 2022-12-25:
Ahh, this fixes the problem: https://github.com/sqfmi/Watchy/pull/193

cromulus, 2022-12-25:
And now this error message: 
```
WARNING: library Rtc_Pcf8563 claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
In file included from /Users/bill/Dropbox/code/Watchy/examples/WatchFaces/7_SEG/7_SEG.ino:2:
/Users/bill/Dropbox/code/Watchy/examples/WatchFaces/7_SEG/settings.h:25:1: error: 'watchySettings' has no non-static data member named 'vibrateOClock'
 };
 ^

exit status 1

Compilation error: 'watchySettings' has no non-static data member named 'vibrateOClock'
```

sqfmi, 2022-12-29:
This should be fixed with the v1.4.5 release, let us know if the issue still persists. Thanks!

closed 2022-12-29 by sqfmi