sqfmi/Watchy #211: How can I intercept Button presses in programming ?

Losspost, 2023-03-05:
Hi, I would love to make a simple test where a message appears on the screen. This then can be affirmed with a button press. I struggle a little bit with it.

I took the `void handleButtonPress(){}` and put it into my `CustomWatchFace` class which is inherited from the `Watchy`. So it should override it but it seems to work differently ? I see that there is a watcher called in the `Watchy::init`. Can I simply remove it from the Watchy Class and put it in my own, or do I have to make the changes in the Watchy class themselves ?

Or is there maybe another recommended way to use the buttons ?

michcan52, 2024-09-11:
Hi, by any chance did you get any progress with this?

peterc, 2025-02-03:
Here's how I did it.

In `Watchy::handleButtonPress()` of the main library, you can add calls to functions like upButton(), downButton() etc. in the non-menu conditionals. Then create empty functions like so:

```
void Watchy::downButton() {
}
```

Put `virtual void downButton();` etc. in the watchy.h next to the drawWatchFace() definition. Then in your own watch subclass you can implement downButton() etc however you wish, like you do with drawWatchFace.