sqfmi/Watchy #215: 'WIDTH_VISIBLE' is not a member of 'WatchyDisplay'

sommerper, 2023-04-18:
Hi!

I'm trying to compile the watch faces from the Arduino 2 IDE.

I'v added the `https://espressif.github.io/arduino-esp32/package_esp32_index.json` to the preferences, selected Watchy as board, and partition scheme as `huge`. Installed the libraries from the manager.

I'm not sure what board version it is but the compiler throws the same error on all the boards anyway.

```
WARNING: library Rtc_Pcf8563 claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
In file included from /home/xxx/Arduino/libraries/Watchy/src/Watchy.h:10,
                 from /home/xxx/Arduino/libraries/Watchy/src/Watchy.cpp:1:
/home/xxx/Arduino/libraries/GxEPD2/src/GxEPD2_BW.h: In instantiation of 'GxEPD2_BW<GxEPD2_Type, page_height>::GxEPD2_BW(GxEPD2_Type) [with GxEPD2_Type = WatchyDisplay; short unsigned int page_height = 200]':
/home/xxx/Arduino/libraries/Watchy/src/Watchy.cpp:5:69:   required from here
/home/xxx/Arduino/libraries/GxEPD2/src/GxEPD2_BW.h:208:134: error: 'WIDTH_VISIBLE' is not a member of 'WatchyDisplay'
     GxEPD2_BW(GxEPD2_Type epd2_instance) : GxEPD2_GFX_BASE_CLASS(GxEPD2_Type::WIDTH_VISIBLE, GxEPD2_Type::HEIGHT), epd2(epd2_instance)
                                                                                                                                      ^

exit status 1

Compilation error: exit status 1
```

Am I missing something?
Thanks!

denics, 2023-04-19:
two days ago it did compile without issues on my MacBookPro M1 but today I get the exact error on my MacPro 6.1 both running OSX and in both cases it was a clear arduino install (I just got my watchy :D )

denics, 2023-04-19:
After some digging I found that the source of the error is in the last version of the https://github.com/ZinggJM/GxEPD2 library 1.5.2 .

As a workaround use version 1.5.1 because in version 1.5.2 they introduced the [WIDTH_VISIBLE](https://github.com/ZinggJM/GxEPD2#version-152) variable.

sommerper, 2023-04-19:
That worked! Thanks for digging into this!

I'm in the same boat as you having just received my Watchy some days ago.

The question is if I should mark this as resolved since it doesn't really fix the issue with the repo...

wjgeorge, 2023-04-22:
is a bug in Watchy, not the external library. They added a static member that is not reflected in Display.h

The watchy Display subsystem is highly coupled to GxEPD2, and needs maintenance whenever anything changes in the base

scottbob09, 2023-04-22:
> That worked! Thanks for digging into this!
> 
> I'm in the same boat as you having just received my Watchy some days ago.
> 
> The question is if I should mark this as resolved since it doesn't really fix the issue with the repo...

I've had mine for a long time... this is just par for the course.

wjgeorge, 2023-04-23:
@scottbob09 close it when its in the Watchy Arduino library (maybe 1.4.7)

AFAIK it's still pending

sqfmi, 2023-04-24:
Thanks for flagging @sommerper and the fix @wjgeorge! Fix pushed in v1.4.7

wjgeorge, 2023-04-26:
@sommerper please test and close if fixed

sommerper, 2023-04-26:
@wjgeorge Yes, sorry for the delay. I'll test tomorrow and update the issue.

Thanks!

sommerper, 2023-05-01:
I can confirm the latest release compiles without any problems in the Arduino editor.

Thank you!

closed 2023-05-01 by sommerper