sqfmi/Watchy #216: Compilation of 7Seg using PlatformIO fails

MichalLebeda, 2023-04-19:
```
Compiling .pio/build/esp32dev/lib96e/Watchy/Watchy.cpp.o
In file included from .pio/libdeps/esp32dev/Watchy/src/WatchyRTC.h:4:0,
                 from .pio/libdeps/esp32dev/Watchy/src/Watchy.h:15,
                 from .pio/libdeps/esp32dev/Watchy/src/Watchy.cpp:1:
.pio/libdeps/esp32dev/Watchy/src/config.h:7:17: note: #pragma message: Please install the latest ESP32 Arduino Core (2.0.5+) and choose Watchy as the target board
 #pragma message "Please install the latest ESP32 Arduino Core (2.0.5+) and choose Watchy as the target board"
                 ^
.pio/libdeps/esp32dev/Watchy/src/config.h:8:17: note: #pragma message: Hardware revision is not defined at the project level, please define in config.h. Defaulting to ARDUINO_WATCHY_V20
 #pragma message "Hardware revision is not defined at the project level, please define in config.h. Defaulting to ARDUINO_WATCHY_V20"
                 ^
In file included from .pio/libdeps/esp32dev/Watchy/src/Watchy.h:10:0,
                 from .pio/libdeps/esp32dev/Watchy/src/Watchy.cpp:1:
.pio/libdeps/esp32dev/GxEPD2/src/GxEPD2_BW.h: In instantiation of 'GxEPD2_BW<GxEPD2_Type, page_height>::GxEPD2_BW(GxEPD2_Type) [with GxEPD2_Type = WatchyDisplay; short unsigned int page_height = 200u]':
.pio/libdeps/esp32dev/Watchy/src/Watchy.cpp:5:69:   required from here
.pio/libdeps/esp32dev/GxEPD2/src/GxEPD2_BW.h:200:134: error: 'WIDTH_VISIBLE' is not a member of 'WatchyDisplay'
     GxEPD2_BW(GxEPD2_Type epd2_instance) : GxEPD2_GFX_BASE_CLASS(GxEPD2_Type::WIDTH_VISIBLE, GxEPD2_Type::HEIGHT), epd2(epd2_instance)
```
platformio.ini:
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = espressif32 @ ~3.5.0
board = esp32dev
framework = arduino
lib_deps =
    sqfmi/Watchy @ 1.4.6 ; Pinned version to ensure we don't pull broken code
    https://github.com/elpaso/Rtc_Pcf8563 ;
    https://github.com/tzapu/WiFiManager.git#v2.0.11-beta ; Pinned for the same reason
lib_ldf_mode = deep+
board_build.partitions = min_spiffs.csv

; upload_speed = 3000000
upload_speed = 115200
; upload_port = /dev/cu.usbserial-MQK8G8
; monitor_port = /dev/cu.usbserial-MQK8G8
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
```
The library has probably changed constant names.

denics, 2023-04-19:
this is the same as in #215 the problem is that you are using the last version of GxEPD2 that does introduce WIDTH_VISIBLE.

denics, 2023-04-21:
if you are using the arduino ide, simply select the library and you will see that you can select a previous version

MichalLebeda, 2023-04-21:
I use vscode and platformio unfortunately

denics, 2023-04-22:
#217 will fix this too

sqfmi, 2023-04-24:
Fix pushed in v1.4.7, please give it a try and see if it's resolved. Thanks!