sqfmi/Watchy #228: Upload "Error 2"

felixscho877, 2023-05-15:
Recently, when flashing via serial connection, I'm getting the well known ESP32-"Error 2". Which means the ESP is not in download-mode. With a normal Board one would now press the EN-Button while downloading. Actually this works also on my Watchy, pulling down the TP0 (GPIO0) to GND. (But of course I don't want do disassemble half of the Watchy every time I want to download a program.)  
I guess Watchy is using the automatic bootloader to get in the bootmode. It seems like this is not working anymore, either due to a HW- or a SW-Fault. 
Has anyone experienced the same problem?
And does someone know why the automatic bootloader fails? Could it be a hardware issue?

sc68cal, 2023-05-24:
I am having the same issue with my newly purchased Watchy. Cannot upload, the watch appears to reboot but does not vibrate when the esptool step fails

sc68cal, 2023-05-24:
The issue is all my motherboard USB ports are USB3.0 and while they do work with other USB 2.0 devices, it does not work with Watchy. Like others on Discord, I had to use a USB 2.0 hub (conveniently part of my monitor) to connect the watch and have everything work.

Techsteps, 2023-06-01:
I have been unable to upload anything on any USB port. I even broke out a 20 year old USB 2.0 hub I had lying around. I get an Error 2 every time. Multiple cables as well.

Mentor1337, 2023-06-14:
I will have to try that. I am getting this weird architecture error as well. I haven't had any luck with Platform or the BLE upload either. I guess I will try 2.0.

felixscho877, 2024-01-30:
For the bootmode-problem, I've found a better a better solution than messing around with wires and pads every time flashing. It also works to stick in the USB-Port just in the right moment, before the flash commands. The time window is about half a second.