sqfmi/Watchy #232: Timezone from OpenWeatherMap has changed, Watchy hasn't.

GuruSR, 2023-05-28:
Line 662:
`          gmtOffset = int(responseObject["timezone"]);`

should be

`         gmtOffset = int(responseObject["timezone"]) / 3600;  // OpenWeatherMap returns seconds from UTC.`

Also, City ID has been discontinued, Geo-Location is preferred.

Sudrien, 2023-06-02:
Geolocation would be nice for sunset and tidal calculation also

Rob58329, 2023-09-11:
In case it's helpfull to anyone:

(1) OpenWeatherMap has always returned gmtOffset in seconds, and this is what Watchy requires/expects, so the original line: `gmtOffset = int(responseObject["timezone"]);`
is correct.

(2) The way I fixed the “City ID has been discontinued, Geo-Location (longitude and latitude) is required” issue is:

(a) Edit “settings.h” to:

`#define OPENWEATHERMAP_URL "http://api.openweathermap.org/data/2.5/weather?lon=-0.12&lat=51.5" `
(or whatever longitude and latitude you want is)

(b) Edit “Watchy.cpp” to:

  ```
      String weatherQueryURL = url + String("&units=") + units +
                               String("&lang=") + lang + String("&appid=") +
                               apiKey; // edited
      http.begin(weatherQueryURL.c_str());
      int httpResponseCode = http.GET();
      if (httpResponseCode == 200) {
        String payload             = http.getString();
        payload=payload.substring(payload.indexOf('{')); // edited/added
        JSONVar responseObject     = JSON.parse(payload);
        currentWeather.temperature = int(responseObject["main"]["temp"]);
        ...
```
(this works as at 11Sept2023)

GuruSR, 2023-09-12:
Instead of using static Longitude and Latitude, you can use the same payload method of reading the JSON from this URL:  http://ip-api.com/json/?fields=lat,lon

And merging the results into the OPENWEATHERMAP_URL by making lon=#1&lat=#2 and replacing #1 & #2 with the results from the above URL after parsong it's JSON.  This is how Watchy GSR does it.

Sudrien, 2024-04-26:
This can be closed as of [99cacd2](https://github.com/sqfmi/Watchy/commit/99cacd26a96f0689926e59e27dfc855cd1849e5e), can't it?