sqfmi/Watchy #235: Timezone incorrect

edent, 2023-06-29:
I'm currently in BST (UTC +1). After its first NTP update, my Watchy thinks it is in UTC -5.

If I check weather.com from a computer on the same LAN, it correctly geolocates me.

Is there a way to fix this? It looks like there's no way to change timezone on the device - so I assume it's something I have to set in the firmware?

TimSmith714, 2023-06-29:
There's a setting in the settings.h file. I can't remember if there is a setting in the watchy interface itself

```
#define GMT_OFFSET_SEC 3600 * -5 //New York is UTC -5 EST, -4 EDT, will be overwritten by weather data
```

I'm in the UK so I've set this to 0 for mine. It does require compiling in the Arduino IDE and updating from a PC.

edent, 2023-06-30:
I'm in the UK and set it to `+1` - that worked. Thanks.

closed 2023-06-30 by edent