sqfmi/Watchy #239: Time stuck to 45:85 2165/25/45

robin-thoni, 2023-10-01:
My Watchy is stuck to 45:85 2165/25/45, even after NTP sync, firmware reset/reflash or setting time manually: https://owncloud.rthoni.com/s/RiGMiG9iss7CrL7

Is there anything wrong with the RTC? What else can I try? The local vendor requested to come to GH first.

THanks

robin-thoni, 2023-10-02:
I forgot to link this picture:
[image]

(This is why I think the RTC is damaged)

sqfmi, 2024-04-02:
@robin-thoni where was this Watchy purchased from? It may be best to return the faulty unit to your vendor for a replacement unit.

robin-thoni, 2024-04-05:
It was purchased from Mouser (France). They already agreed to replace the unit, but they are out of stock for over 6 months now...

229311JK, 2024-05-01:
> @robin-thoni where was this Watchy purchased from? It may be best to return the faulty unit to your vendor for a replacement unit.

Did you get anywhere w/this?  Looks like a software issue to me.  On hardware, maybe the E-paper ribbon cable isn't properly connected...  Hey Robin Thoni, what does core debug tell you?  What info you need should manifest there in serial output....

robin-thoni, 2024-05-08:
This is what I have:
```
[C][PCF8563:013]: Setting up PCF8563...
[E][PCF8563:086]: Can't read I2C data.
[E][component:119]: Component pcf8563.time was marked as failed.
[E][component:164]: Component pcf8563.time set Error flag: unspecified
...
[C][PCF8563:022]: PCF8563:
[C][PCF8563:023]:   Address: 0xA3
[E][PCF8563:025]: Communication with PCF8563 failed!
[C][PCF8563:027]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[E][component:082]:   Component pcf8563.time is marked FAILED
```

The RTC just does not reply. This why I get `<UNKNOWN>`.

229311JK, 2024-05-09:
What's up Mr. Thoni; 1st, go here:  https://watchy.sqfmi.com/docs/hardware to checkbox 21.  2nd. since, your Mouser France, verify board revision w/BOM.  3rd, download a copy of the pdf file WatchySchematic.   Blow it up, find the RTC in the bottom right.  4th, use/get an Oscilloscope my friend to scope this little misbehaving RTC gangster chip.  5th, cut and paste your errors in a good search engine for something like this:  https://www.reddit.com/r/AskElectronics/comments/1939wrd/pcf8563_rtc_not_working_in_my_custom_pcb/?rdt=42325.  6 and finally, come back here & tell me more please.  Truly, I hope this throws some light on your situation...

robin-thoni, 2024-05-09:
I don't have the required hardware to scope a PCB w/ such little space/pin space. However, I just realized this:
```
[I][i2c.arduino:096]: Results from i2c bus scan:
[I][i2c.arduino:098]: Found no i2c devices!
```
Meaning the BMA423 does not seem to respond either. Maybe the problem is w/ the i2c bus itself (short circuit, or something), or the esp chip.

According to [this](https://github.com/sqfmi/Watchy/blame/master/src/Watchy.cpp#L723), a v3 seems to be coming. Maybe, I'm just gonna ask a refund for this unit and wait for the v3

229311JK, 2024-05-10:
Hey, this may be little but its progress Sir?  Now, the BMA423 input helps...  1st, please look@ the BMA423 sensor data sheet.  2nd, can your verify the e-paper screen works?  3rd, what flashware are-u using?  4th, I'd suggest erasing it w/the Espressif flash tool.  But that's your decision....  5th, open the case & give me a picture of your ribbon cable connection please Mr. Thoni.  6th and finally, find a scope to borrow or make a DDS one.  Other business, loved your repos.  If you got all that your resourceful enough to get a scope.  By the way, someone tried to mag spoof me last night....   Forgot, debug would tell you something about a short or issue w/the ESP32.  Thanks for pointing out that comment too.  The esp32 S3's & C series are nice...