sqfmi/Watchy #244: Watchy hour keeps resetting

ckirtley, 2024-03-20:
I just bought a Watchy, and noticed the time keeps resetting to "00" (minutes are not affected). I have change the face through Arduino IDE but it was happening before that with the default face. If I correct it, it stays good for a few hours then flips back to 00.

ckirtley, 2024-03-21:
> I just bought a Watchy, and noticed the time keeps resetting to "21" (minutes are not affected). I have change the face through Arduino IDE but it was happening before that with the default face. If I correct it, it stays good for a few hours then flips back to 21.

ckirtley, 2024-03-21:
I just realised the city is set as NYC and time is overwritten by the weather function!

sqfmi, 2024-04-02:
Yes, the timezone is overwritten by the weather API currently, will look into making it optional in the future. Apologies for the confusion!

closed 2024-04-02 by sqfmi