sqfmi/Watchy #248: Compatible E-paper panel sourcing resolved>!<

229311JK, 2024-04-28:
Compatible 1.54" SPI Monochrome 200x200 Pixel E-paper replacement can be found @the hyperlink below...
   
[https://www.seeedstudio.com/1-54-Monochrome-ePaper-Display-with-200x200-Pixels-p-5776.html](url)

DarkZeros, 2024-05-03:
Do you think this one is better than the default one, or is it the same? Is it needed to change the code at all?

229311JK, 2024-05-03:
> Do you think this one is better than the default one, or is it the same? Is it needed to change the code at all?

Personally, yes & it works great so far; Need to change SQFMI WATCHY code yes.  Your code of course is a different beast...  Thanks 4-sharing, didn't know bout it till now...

229311JK, 2024-05-03:
Need a hand w/IDF Arduino core get in touch...

229311JK, 2024-05-10:
> Do you think this one is better than the default one, or is it the same? Is it needed to change the code at all?

Did you try-2 get 1-yet?  Noticed a number of them bought up lately.....  Also, please see the above comment too...

DarkZeros, 2024-05-10:
I didn't buy it since I was checking the documents attached and looked like it is the same controller and display (specs are exactly the same), maybe marketed by another name?

If you need help with something let me know, I am working on a side project to Watchy on my spare time.

229311JK, 2024-05-14:
> I didn't buy it since I was checking the documents attached and looked like it is the same controller and display (specs are exactly the same), maybe marketed by another name?
> 
> If you need help with something let me know, I am working on a side project to Watchy on my spare time.

Yep, I spent hours going over datasheets 4-E-paper panels.... This version panel has a different PN & is more responsive than the OEM.  On help, thank you will be in touch....