sqfmi/Watchy #250: esp32:esp32@3.0.0 breaks build with 1.4.10

Sudrien, 2024-05-31:
Setting up Arduino IDE on a reformatted computer. 

Find https://github.com/espressif/arduino-esp32/tree/3.0.0 was tagged 4 days ago, as of writing.

The answer as of Watchy 1.4.10 is "Use 2.0.x of the esp32 by Expressif Systems library (currently 2.0.17)"

If you don't, the dump looks somthing like this

<pre>
> WARNING: library Rtc_Pcf8563 claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
> ~/Arduino/libraries/Watchy/src/Display.cpp: In static member function 'static void WatchyDisplay::busyCallback(const void*)':
> ~/Arduino/libraries/Watchy/src/Display.cpp:23:3: error: 'gpio_wakeup_enable' was not declared in this scope
>    23 |   gpio_wakeup_enable((gpio_num_t)DISPLAY_BUSY, GPIO_INTR_LOW_LEVEL);
>       |   ^~~~~~~~~~~~~~~~~~
> ~/Arduino/libraries/Watchy/src/BLE.cpp: In member function 'virtual void otaCallback::onWrite(BLECharacteristic*)':
> ~/Arduino/libraries/Watchy/src/BLE.cpp:41:49: error: conversion from 'String' to non-scalar type 'std::string' {aka 'std::__cxx11::basic_string<char>'} requested
>    41 |   std::string rxData = pCharacteristic->getValue();
>       |                        ~~~~~~~~~~~~~~~~~~~~~~~~~^~
> In file included from ~/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/esp32-hal-gpio.h:29,
>                  from ~/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/esp32-hal.h:75,
>                  from ~/.arduino15/packages/esp32/hardware/esp32/3.0.0/cores/esp32/Arduino.h:36,
>                  from ~/Arduino/libraries/Watchy/src/Watchy.h:4,
>                  from ~/Arduino/libraries/Watchy/src/Watchy.cpp:1:
> ~/Arduino/libraries/Watchy/src/Watchy.cpp: In member function 'void Watchy::deepSleep()':
> ~/.arduino15/packages/esp32/hardware/esp32/3.0.0/variants/watchy/pins_arduino.h:46:23: error: 'GPIO_SEL_26' was not declared in this scope; did you mean 'GPIO_NUM_26'?
>    46 | #define MENU_BTN_MASK GPIO_SEL_26
>       |                       ^~~~~~~~~~~
> ~/.arduino15/packages/esp32/hardware/esp32/3.0.0/variants/watchy/pins_arduino.h:50:23: note: in expansion of macro 'MENU_BTN_MASK'
>    50 | #define BTN_PIN_MASK  MENU_BTN_MASK | BACK_BTN_MASK | UP_BTN_MASK | DOWN_BTN_MASK
>       |                       ^~~~~~~~~~~~~
> ~/Arduino/libraries/Watchy/src/Watchy.cpp:84:7: note: in expansion of macro 'BTN_PIN_MASK'
>    84 |       BTN_PIN_MASK,
>       |       ^~~~~~~~~~~~
> ~/.arduino15/packages/esp32/hardware/esp32/3.0.0/variants/watchy/pins_arduino.h:47:23: error: 'GPIO_SEL_25' was not declared in this scope; did you mean 'GPIO_NUM_25'?
>    47 | #define BACK_BTN_MASK GPIO_SEL_25
>       |                       ^~~~~~~~~~~
> ~/.arduino15/packages/esp32/hardware/esp32/3.0.0/variants/watchy/pins_arduino.h:50:39: note: in expansion of macro 'BACK_BTN_MASK'
>    50 | #define BTN_PIN_MASK  MENU_BTN_MASK | BACK_BTN_MASK | UP_BTN_MASK | DOWN_BTN_MASK
>       |                                       ^~~~~~~~~~~~~
> ~/Arduino/libraries/Watchy/src/Watchy.cpp:84:7: note: in expansion of macro 'BTN_PIN_MASK'
>    84 |       BTN_PIN_MASK,
>       |       ^~~~~~~~~~~~
> ~/.arduino15/packages/esp32/hardware/esp32/3.0.0/variants/watchy/pins_arduino.h:32:21: error: 'GPIO_SEL_32' was not declared in this scope; did you mean 'GPIO_NUM_32'?
>    32 | #define UP_BTN_MASK GPIO_SEL_32
>       |                     ^~~~~~~~~~~
> ~/.arduino15/packages/esp32/hardware/esp32/3.0.0/variants/watchy/pins_arduino.h:50:55: note: in expansion of macro 'UP_BTN_MASK'
>    50 | #define BTN_PIN_MASK  MENU_BTN_MASK | BACK_BTN_MASK | UP_BTN_MASK | DOWN_BTN_MASK
>       |                                                       ^~~~~~~~~~~
> ~/Arduino/libraries/Watchy/src/Watchy.cpp:84:7: note: in expansion of macro 'BTN_PIN_MASK'
>    84 |       BTN_PIN_MASK,
>       |       ^~~~~~~~~~~~
> ~/.arduino15/packages/esp32/hardware/esp32/3.0.0/variants/watchy/pins_arduino.h:48:23: error: 'GPIO_SEL_4' was not declared in this scope; did you mean 'GPIO_NUM_4'?
>    48 | #define DOWN_BTN_MASK GPIO_SEL_4
>       |                       ^~~~~~~~~~
> ~/.arduino15/packages/esp32/hardware/esp32/3.0.0/variants/watchy/pins_arduino.h:50:69: note: in expansion of macro 'DOWN_BTN_MASK'
>    50 | #define BTN_PIN_MASK  MENU_BTN_MASK | BACK_BTN_MASK | UP_BTN_MASK | DOWN_BTN_MASK
>       |                                                                     ^~~~~~~~~~~~~
> ~/Arduino/libraries/Watchy/src/Watchy.cpp:84:7: note: in expansion of macro 'BTN_PIN_MASK'
>    84 |       BTN_PIN_MASK,
>       |       ^~~~~~~~~~~~
> ~/Arduino/libraries/Watchy/src/Watchy.cpp: In member function 'virtual void Watchy::handleButtonPress()':
> ~/.arduino15/packages/esp32/hardware/esp32/3.0.0/variants/watchy/pins_arduino.h:46:23: error: 'GPIO_SEL_26' was not declared in this scope; did you mean 'GPIO_NUM_26'?
>    46 | #define MENU_BTN_MASK GPIO_SEL_26
>       |                       ^~~~~~~~~~~
> ~/Arduino/libraries/Watchy/src/Watchy.cpp:92:19: note: in expansion of macro 'MENU_BTN_MASK'
>    92 |   if (wakeupBit & MENU_BTN_MASK) {
>       |                   ^~~~~~~~~~~~~
> ~/.arduino15/packages/esp32/hardware/esp32/3.0.0/variants/watchy/pins_arduino.h:47:23: error: 'GPIO_SEL_25' was not declared in this scope; did you mean 'GPIO_NUM_25'?
>    47 | #define BACK_BTN_MASK GPIO_SEL_25
>       |                       ^~~~~~~~~~~
> ~/Arduino/libraries/Watchy/src/Watchy.cpp:128:24: note: in expansion of macro 'BACK_BTN_MASK'
>   128 |   else if (wakeupBit & BACK_BTN_MASK) {
>       |                        ^~~~~~~~~~~~~
> ~/.arduino15/packages/esp32/hardware/esp32/3.0.0/variants/watchy/pins_arduino.h:32:21: error: 'GPIO_SEL_32' was not declared in this scope; did you mean 'GPIO_NUM_32'?
>    32 | #define UP_BTN_MASK GPIO_SEL_32
>       |                     ^~~~~~~~~~~
> ~/Arduino/libraries/Watchy/src/Watchy.cpp:141:24: note: in expansion of macro 'UP_BTN_MASK'
>   141 |   else if (wakeupBit & UP_BTN_MASK) {
>       |                        ^~~~~~~~~~~
> ~/.arduino15/packages/esp32/hardware/esp32/3.0.0/variants/watchy/pins_arduino.h:48:23: error: 'GPIO_SEL_4' was not declared in this scope; did you mean 'GPIO_NUM_4'?
>    48 | #define DOWN_BTN_MASK GPIO_SEL_4
>       |                       ^~~~~~~~~~
> ~/Arduino/libraries/Watchy/src/Watchy.cpp:153:24: note: in expansion of macro 'DOWN_BTN_MASK'
>   153 |   else if (wakeupBit & DOWN_BTN_MASK) {
>       |                        ^~~~~~~~~~~~~
> 
> exit status 1
> 
> Compilation error: exit status 1
</pre>

229311JK, 2024-06-03:
Hmmmm....  Year 1 Watchy wear/tear operation complete;  Have noticed float charge affect pin performance.  Swore I used the dev version of 3.0, had to modify my warnings.  Maybe I'm wrong, but here's the variant h-file....
[image]

229311JK, 2024-06-09:
External wake up ext1 problem #9809
[image]

sqfmi, 2024-07-02:
Should be fixed now in the latest release, let us know if there are any issues with 3.0. Thanks!

229311JK, 2024-07-06:
> Should be fixed now in the latest release, let us know if there are any issues with 3.0. Thanks!

I need an avenue 2-contact you;  Really just want-2 help you & darkzeros out cause you got a great product/platform.   Also become a contributor here-2.  It really sucks you had to go through all sourcing issues....  Like I said, might b-able 2 help.  ; )

closed 2024-07-07 by sqfmi