sqfmi/Watchy #252: How to update ESP32-S3 model (from CrowdSupply campaign)

Utopiah, 2024-07-01:
Hi, just received my Watchy and it works well. Unfortunately the timezone is on GMT -5 and I'm on GMT +2. I didn't find a way to change that via the interface on the watch. I did learn via https://github.com/sqfmi/Watchy/issues/235 that I have to compile the firmware. I'm a bit lazy so I thought I'd try first via https://watchy.sqfmi.com/watchfaces using Chromium but it didn't work.

I thought maybe my USB cable was wrong so I tried `espflash board-info` which did show me 

```
Chip type:         esp32s3 (revision v0.2)
Crystal frequency: 40MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       48:ca:43:56:cb:7c
```

from `Serial port: '/dev/ttyACM0'` and thus both the watch and cable are working.

Consequently I'm wondering what's the status of support for this model. Is it a new one? Can I follow https://watchy.sqfmi.com/docs/getting-started or is it for V2.0 which might be an older version?

sqfmi, 2024-07-02:
Hey there, it looks like you have the latest hardware which is V3 and uses the ESP32-S3. You will need to update to the latest arduino-esp32 (v3.0.2) and the latest Watchy library (v1.4.12).

In the Arduino IDE, select the following options:

- Board: ESP32S3 Dev Module
- Flash Size: 8MB
- Partition Scheme: 8MB with Spiffs

To enter bootloader mode:
- Hold both top buttons (back and up) for 4 seconds or more, then release the back button first, then the up button

To reset Watchy:
- Hold both top buttons (back and up) for 4 seconds or more, then release the up button first, then the back button

Enter bootloader mode, select the serial port, upload sketch, reset Watchy

We'll be updating the official docs shortly.

Utopiah, 2024-07-03:
Thanks for the clarification. It did seem to work (verifying and uploading) and worked as I am now on the right timezone and can thus use NTP.

When I do use "Update Firmware" though and go on Chromium on https://watchy.sqfmi.com/watchfaces and select another watchface, e.g `7_SEG_LIGHT` it does seem to work yet I don't get the new watchface, even after reset. Is it expected? If I do it via Arduino and change `DARKMODE` in `Watchy_7_SEG.cpp` though it works.

sqfmi, 2024-07-05:
The "Update Firmware" functionality is deprecated and will be updated with a new version with OTA soon. For now, please select example watchfaces in the Arduino IDE and upload them over USB.

closed 2024-07-05 by sqfmi