sqfmi/Watchy #253: A fatal error occurred: This chip is ESP32-S3 not ESP32. Wrong --chip argument?

celevra, 2024-07-02:
Hi,

i've a new watchy and try to upgrade it to the newest firmware and change the weather location to get the right timezone.
i've installed esp32 v2.017, Arduino ESP32 Boards v2.0.13 and Watchy Lib 1.4.10 from the library manager.
i've selected the watchy board and Hardware v2.0
When i try to upload the compiled code i get the error:

A fatal error occurred: This chip is ESP32-S3 not ESP32. Wrong --chip argument?

but i have no glue how to change the chip argument, can you guide me in the right direction?

regards

celevra, 2024-07-02:
could be that it is a duplicate of https://github.com/sqfmi/Watchy/issues/252

```
$ ./espflash board-info
Chip type:         esp32s3 (revision v0.2)
Crystal frequency: 40 MHz
Flash size:        8MB
Features:          WiFi, BLE
MAC address:       48:ca:43:54:2a:20
```

sqfmi, 2024-07-02:
Hey there, it looks like you have the latest hardware which is V3 and uses the ESP32-S3. You will need to update to the latest arduino-esp32 (v3.0.2) and the latest Watchy library (v1.4.12). We'll be updating the official docs shortly.

celevra, 2024-07-03:
thanks for the fast answer
when i update esp32 to v3.0.2 and watchy to 1.4.12 and open examples 7_seg.ino and click on upload i get compilation errors:

```
WARNING: library Rtc_Pcf8563 claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
In file included from /home/celevra/Arduino/libraries/Watchy/src/WatchyRTC.h:4,
                 from /home/celevra/Arduino/libraries/Watchy/src/WatchyRTC.cpp:1:
/home/celevra/Arduino/libraries/Watchy/src/config.h:49:22: error: expected unqualified-id before numeric constant
   49 | #define MENU_BTN_PIN 26
      |                      ^~
/home/celevra/.arduino15/packages/esp32/hardware/esp32/3.0.2/variants/watchy/pins_arduino.h:17:22: note: in expansion of macro 'MENU_BTN_PIN'
   17 | static const uint8_t MENU_BTN_PIN = 26;
      |                      ^~~~~~~~~~~~
/home/celevra/Arduino/libraries/Watchy/src/config.h:50:22: error: expected unqualified-id before numeric constant
   50 | #define BACK_BTN_PIN 25
      |                      ^~
/home/celevra/.arduino15/packages/esp32/hardware/esp32/3.0.2/variants/watchy/pins_arduino.h:18:22: note: in expansion of macro 'BACK_BTN_PIN'
   18 | static const uint8_t BACK_BTN_PIN = 25;
      |                      ^~~~~~~~~~~~
/home/celevra/Arduino/libraries/Watchy/src/config.h:51:22: error: expected unqualified-id before numeric constant
   51 | #define DOWN_BTN_PIN 4
      |                      ^
/home/celevra/.arduino15/packages/esp32/hardware/esp32/3.0.2/variants/watchy/pins_arduino.h:19:22: note: in expansion of macro 'DOWN_BTN_PIN'
   19 | static const uint8_t DOWN_BTN_PIN = 4;
      |                      ^~~~~~~~~~~~
/home/celevra/Arduino/libraries/Watchy/src/config.h:52:20: error: expected unqualified-id before numeric constant
   52 | #define DISPLAY_CS 5
      |                    ^
/home/celevra/.arduino15/packages/esp32/hardware/esp32/3.0.2/variants/watchy/pins_arduino.h:20:22: note: in expansion of macro 'DISPLAY_CS'
   20 | static const uint8_t DISPLAY_CS = 5;
      |                      ^~~~~~~~~~
/home/celevra/Arduino/libraries/Watchy/src/config.h:53:21: error: expected unqualified-id before numeric constant
   53 | #define DISPLAY_RES 9
      |                     ^
/home/celevra/.arduino15/packages/esp32/hardware/esp32/3.0.2/variants/watchy/pins_arduino.h:21:22: note: in expansion of macro 'DISPLAY_RES'
   21 | static const uint8_t DISPLAY_RES = 9;
      |                      ^~~~~~~~~~~
/home/celevra/Arduino/libraries/Watchy/src/config.h:54:20: error: expected unqualified-id before numeric constant
   54 | #define DISPLAY_DC 10
      |                    ^~
/home/celevra/.arduino15/packages/esp32/hardware/esp32/3.0.2/variants/watchy/pins_arduino.h:22:22: note: in expansion of macro 'DISPLAY_DC'
   22 | static const uint8_t DISPLAY_DC = 10;
      |                      ^~~~~~~~~~
/home/celevra/Arduino/libraries/Watchy/src/config.h:55:22: error: expected unqualified-id before numeric constant
   55 | #define DISPLAY_BUSY 19
      |                      ^~
/home/celevra/.arduino15/packages/esp32/hardware/esp32/3.0.2/variants/watchy/pins_arduino.h:23:22: note: in expansion of macro 'DISPLAY_BUSY'
   23 | static const uint8_t DISPLAY_BUSY = 19;
      |                      ^~~~~~~~~~~~
/home/celevra/Arduino/libraries/Watchy/src/config.h:56:23: error: expected unqualified-id before numeric constant
   56 | #define ACC_INT_1_PIN 14
      |                       ^~
/home/celevra/.arduino15/packages/esp32/hardware/esp32/3.0.2/variants/watchy/pins_arduino.h:24:22: note: in expansion of macro 'ACC_INT_1_PIN'
   24 | static const uint8_t ACC_INT_1_PIN = 14;
      |                      ^~~~~~~~~~~~~
/home/celevra/Arduino/libraries/Watchy/src/config.h:57:23: error: expected unqualified-id before numeric constant
   57 | #define ACC_INT_2_PIN 12
      |                       ^~
/home/celevra/.arduino15/packages/esp32/hardware/esp32/3.0.2/variants/watchy/pins_arduino.h:25:22: note: in expansion of macro 'ACC_INT_2_PIN'
   25 | static const uint8_t ACC_INT_2_PIN = 12;
      |                      ^~~~~~~~~~~~~
/home/celevra/Arduino/libraries/Watchy/src/config.h:58:23: error: expected unqualified-id before numeric constant
   58 | #define VIB_MOTOR_PIN 13
      |                       ^~
/home/celevra/.arduino15/packages/esp32/hardware/esp32/3.0.2/variants/watchy/pins_arduino.h:26:22: note: in expansion of macro 'VIB_MOTOR_PIN'
   26 | static const uint8_t VIB_MOTOR_PIN = 13;
      |                      ^~~~~~~~~~~~~
/home/celevra/Arduino/libraries/Watchy/src/config.h:59:21: error: expected unqualified-id before numeric constant
   59 | #define RTC_INT_PIN 27
      |                     ^~
/home/celevra/.arduino15/packages/esp32/hardware/esp32/3.0.2/variants/watchy/pins_arduino.h:27:22: note: in expansion of macro 'RTC_INT_PIN'
   27 | static const uint8_t RTC_INT_PIN = 27;
      |                      ^~~~~~~~~~~
/home/celevra/Arduino/libraries/Watchy/src/config.h:72:24: error: expected unqualified-id before numeric constant
   72 |     #define UP_BTN_PIN 35
      |                        ^~
/home/celevra/.arduino15/packages/esp32/hardware/esp32/3.0.2/variants/watchy/pins_arduino.h:40:22: note: in expansion of macro 'UP_BTN_PIN'
   40 | static const uint8_t UP_BTN_PIN = 35;
      |                      ^~~~~~~~~~
/home/celevra/Arduino/libraries/Watchy/src/config.h:73:26: error: expected unqualified-id before numeric constant
   73 |     #define BATT_ADC_PIN 34
      |                          ^~
/home/celevra/.arduino15/packages/esp32/hardware/esp32/3.0.2/variants/watchy/pins_arduino.h:41:22: note: in expansion of macro 'BATT_ADC_PIN'
   41 | static const uint8_t BATT_ADC_PIN = 34;
      |                      ^~~~~~~~~~~~

exit status 1

Compilation error: exit status 1
```

seekermarcel, 2024-07-03:
can confirm this error. I have the exact same one under the same circumstances

Utopiah, 2024-07-03:
You're probably still compiling against the older devboard. Happened to me too then once I "properly" switched (unfortunately not sure what I did initially wrong) it did work without any error, just a single warning. I did verify then upload and had it working, cf https://github.com/sqfmi/Watchy/issues/252#issuecomment-2207126789

thesysop, 2024-07-04:
I am also getting this error using the Arduino IDE 2.3.3 with esp32 version 3.0.2 installed and Watchy 1.4.12 on the latest crowdsupply hardware. The examples fail compilation with the errors seen above. One thing to note is in the "Board Revision" settings under tools there is no Watchy V3.0, only v1.0, v1.5 and v2.0.

celevra, 2024-07-04:
This works:
In the Arduino IDE, select the following options:

Board: ESP32S3 Dev Module
Flash Size: 8MB
Partition Scheme: 8MB with Spiffs
To enter bootloader mode:

Hold both top buttons (back and up) for 4 seconds or more, then release the back button first, then the up button
To reset Watchy:

Hold both top buttons (back and up) for 4 seconds or more, then release the up button first, then the back button
Enter bootloader mode, select the serial port, upload sketch, reset Watchy

thesysop, 2024-07-04:
@celevra That works! I was able to both verify and upload the seven segment example. Thanks for sharing.

closed 2024-07-05 by sqfmi