sqfmi/Watchy #257: Clock time not updating/screen not updating

Baumart, 2024-07-05:
In the newst build 1.4.13:
Time is running BUT the screen is not updating. Only manuel refresh helps but after 1min, the screen-time is out off sync.

Baumart, 2024-07-05:
in the ver 1.4.12 the "Bug" is not there

Schokonoko, 2024-07-05:
Experienced the same thing. Got my watchy yesterday worked fine out of the box adjusted the firmware settings to reflect my Timezone. Watchy then only updated time when entering the menu and exiting. Found this issue report and rolled back to version 1.4.12. now its working as expected again

sqfmi, 2024-07-06:
@Baumart @Schokonoko what is the version of arduino-esp32? Currently only v2.0.17 is supported. Can you also try updating Watchy library to v1.4.14?

Baumart, 2024-07-06:
Watchy ver. 1.4.14 with eps23 v2.0.17 is okay, i think i accidentally used v3.0.2 ... ok lol

closed 2024-07-06 by Baumart