sqfmi/Watchy #259: After Verfy and Upload on Arduino IDE Watchy freezed on menu screen.

Inqb8tr-jp, 2024-07-08:
Hi,
I received Watchy on July 1st.
 
After following your instructions, Watchy has been frozen on the menu screen.
https://github.com/sqfmi/Watchy/issues/252#issuecomment-2204482586

Resetting sometimes makes vibrations but there is no exit from the menu screen.

sqfmi, 2024-07-09:
Hello, we have updated the instructions here https://watchy.sqfmi.com/docs/getting-started/#arduino-setup

Can you try again using arduino-esp32 v2.0.17 and Watchy Library v1.4.14?

Inqb8tr-jp, 2024-07-09:
Thanks for your reply.

I reupdated the watchy but no good.

I can enter the bootstrap mode.
The reset makes vibrations but no screen refresh.
The screen is blacker like there is no battery.

I tried 7SEG and BASIC watch faces w/wo GMT_OFFSET_SEC modifications but all the same result with no screen refresh.
I replaced the USB cable with another one but got the same results.

My environment is the following.
* esp32 ver 2.0.17
* watchy ver 1.4.14
* SmallRTC ver 2.3.7
* Rtc_Pcf8563 ver 1.0.3
* GxEPD2 ver 1.5.7

sqfmi, 2024-07-09:
Thank you for the details, can you show us your assembly of Watchy? Front and back picture would be helpful. Thanks!

Inqb8tr-jp, 2024-07-09:
Sorry. I can't understand what the assembly means because I'm not a native English speaker.
Do the pictures attached below fulfill your request?
[image]
[image]

sqfmi, 2024-07-10:
Thank you, can you also open the case, and share photo of the front and back of the PCB?

Inqb8tr-jp, 2024-07-10:
Thank you for your support.
After opening the case, everything works fine.
Maybe it was a bad contact.
Thank you.

sqfmi, 2024-07-10:
No problem, make sure the ribbon cable is inserted fully like so:

[image]

Check out our [docs](https://watchy.sqfmi.com/docs/getting-started) and let us know if you have any other questions. Thanks!