sqfmi/Watchy #260: Failed uploading

pm4808, 2024-07-10:
After setting up arduino IDE as per instructions (July 9) I tried uploading the 7Seg example and recieved the following:

A fatal error occurred: This chip is ESP32 not ESP32-S3. Wrong --chip argument?
Failed uploading: uploading error: exit status 2

The watch was then completely unresponsive - not even reset worked, Watch now responsive again.

LibVer: 1.4.10
Rev: V20

vonshednob, 2024-07-11:
I ran into the same problem and don't have an answer yet. Issue #252 suggests to use `esp32:esp32:esp32s3` board instead of `esp32:esp32:watchy`.

However when compiling it with that option the program storage space is used to 132% (7seg example).

Long story short: I'd be very interested in a solution to this, too.

sqfmi, 2024-07-13:
Please follow the getting started guide here and let us know if there are still any issues: https://watchy.sqfmi.com/docs/getting-started.

To compile and upload in Arduino IDE, make sure you update to the latest Watchy library (v1.4.14) and select the arduino-esp32 (v2.0.17) board package. In the Arduino IDE, select the following options:

Board: ESP32S3 Dev Module
Flash Size: 8MB
Partition Scheme: 8MB with Spiffs

To enter bootloader mode:
Hold both top buttons (back and up) for 4 seconds or more, then release the back button first, then the up button

To reset Watchy:
Hold both top buttons (back and up) for 4 seconds or more, then release the up button first, then the back button

Enter bootloader mode, select the serial port, upload sketch, reset Watchy

vonshednob, 2024-07-13:
I checked and I have watchy library v1.4.14 installed (and used) and arduino-esp32 v2.0.17 installed (and in use).

When running the compile step for the 7_SEG example, this happens:

```
$ arduino-cli compile -b esp32:esp32:esp32s3 --build-property build.flash_size=8MB --build-property build.partitions=default_8MB --clean .
WARNING: library Rtc_Pcf8563 claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
Sketch uses 1730173 bytes (132%) of program storage space. Maximum is 1310720 bytes.
Global variables use 73708 bytes (22%) of dynamic memory, leaving 253972 bytes for local variables. Maximum is 327680 bytes.
```

I know to ignore the warning, but the 132% of program storage usage stops the build.

Building instead with `-b esp32:esp32:watchy` works, but then the upload has to be done using `esptool.py` directly, selecting the `esp32s3` board and an additional `--force` (because of course board and build are not consistent in that regard).

Any advice?

sqfmi, 2024-07-13:
@vonshednob try this command instead:

```arduino-cli compile -b esp32:esp32:esp32s3 --board-options FlashSize=8M,PartitionScheme=default_8MB --clean .```

vonshednob, 2024-07-14:
@sqfmi Thank you, this solves the problem for me!

closed 2024-07-14 by sqfmi