sqfmi/Watchy #261: can't upload firmware on linux

paolorussian, 2024-07-12:
Hello, uploading on mint (debian) doesn't work:
```
Serial port /dev/ttyS0
Connecting.....................................An error occurred while uploading the sketch
```
.

A fatal error occurred: Failed to connect to ESP32: No serial data received.

Dmesg says
```
[175063.351198] usb 1-2: new full-speed USB device number 65 using xhci_hcd
[175063.487262] usb 1-2: device descriptor read/64, error -71
[175089.786472] usb 1-9: USB disconnect, device number 44
```

lsusb | grep Silicon returns an empty prompt.

I already tried following the "CP210x USB to UART Bridge" thing but the page but first I think I already have them, secondly the download page doesn't contain any download for linux, it just points to kernel.org:

```
Downloads for Windows, Macintosh, Linux and Android below.
*Note: The Linux 3.x.x and 4.x.x version of the driver is maintained in the current Linux 3.x.x and 4.x.x tree at [www.kernel.org](http://www.kernel.org/).
```

I also have a windows machine that I turned on just for uploading a watchy firmware but on windows my arduino ide doesn't offer "watchy" as a board, even if the json url is configured in preferences.

I tried every usb cable I have at home, tried restarting the pc, also tried 2 different watchy watches, to exclude the possibility of faulty hardware.

Is it possible to upload watchy firmwares on linux anyway? Did anyone succed in this before?

sqfmi, 2024-07-13:
Hello, can you provide a link to the documentation you are using?

davidhaley, 2024-07-14:
I ran into similar issues, submitted a support request to mouser last week (included details below), and just received another watch.

```
Arch Linux
6.9.7-arch1-1
```

---

- The screen flickers from white background and black text, to black background and white text, every time the 'page' changes (i.e. visiting about from settings, going back to the main watch screen, etc)
- I cannot connect to the device at all using Arduino IDE or Arduino CLI
  - I followed the instructions closely to install the ESP32 board and Watchy libraries
  - I loaded the cp210x kernel module for the usb serial driver
  - I tried 3 different (data-capable) micro-usb cables (including high-quality cables)
  - I tried multiple USB host bus devices such as front ports, rear ports, an Anker hub

 
When plugging the device into my machine directly, it continuously tries to connect but disconnects.
 
dmesg output:
```
[ 7150.950404] usb 1-10.3.4.2: new full-speed USB device number 42 using xhci_hcd
[ 7151.042105] usb 1-10.3.4.2: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[ 7151.042117] usb 1-10.3.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 7151.042122] usb 1-10.3.4.2: Product: USB JTAG/serial debug unit
[ 7151.042126] usb 1-10.3.4.2: Manufacturer: Espressif
[ 7151.042129] usb 1-10.3.4.2: SerialNumber: 48:CA:43:54:22:24
[ 7151.056553] cdc_acm 1-10.3.4.2:1.0: ttyACM0: USB ACM device
[ 7155.942292] usb 1-10.3.4.2: USB disconnect, device number 42
[ 7156.150139] usb 1-10.3.4.2: new full-speed USB device number 43 using xhci_hcd
[ 7156.223515] usb 1-10.3.4.2: device descriptor read/64, error -32
[ 7156.421903] usb 1-10.3.4.2: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[ 7156.421916] usb 1-10.3.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 7156.421921] usb 1-10.3.4.2: Product: USB JTAG/serial debug unit
[ 7156.421924] usb 1-10.3.4.2: Manufacturer: Espressif
[ 7156.421927] usb 1-10.3.4.2: SerialNumber: 48:CA:43:54:22:24
[ 7156.436971] cdc_acm 1-10.3.4.2:1.0: ttyACM0: USB ACM device
[ 7161.317118] usb 1-10.3.4.2: USB disconnect, device number 43
[ 7161.523157] usb 1-10.3.4.2: new full-speed USB device number 44 using xhci_hcd
[ 7161.596640] usb 1-10.3.4.2: device descriptor read/64, error -32
[ 7161.776558] usb 1-10.3.4.2: device descriptor read/64, error -32
[ 7161.956440] usb 1-10.3.4.2: new full-speed USB device number 45 using xhci_hcd
[ 7162.048136] usb 1-10.3.4.2: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[ 7162.048152] usb 1-10.3.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 7162.048158] usb 1-10.3.4.2: Product: USB JTAG/serial debug unit
[ 7162.048163] usb 1-10.3.4.2: Manufacturer: Espressif
[ 7162.048167] usb 1-10.3.4.2: SerialNumber: 48:CA:43:54:22:24
[ 7162.062735] cdc_acm 1-10.3.4.2:1.0: ttyACM0: USB ACM device
[ 7166.949055] usb 1-10.3.4.2: USB disconnect, device number 45
[ 7167.156222] usb 1-10.3.4.2: new full-speed USB device number 46 using xhci_hcd
[ 7167.229631] usb 1-10.3.4.2: device descriptor read/64, error -32
[ 7167.427780] usb 1-10.3.4.2: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[ 7167.427796] usb 1-10.3.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 7167.427803] usb 1-10.3.4.2: Product: USB JTAG/serial debug unit
[ 7167.427807] usb 1-10.3.4.2: Manufacturer: Espressif
[ 7167.427811] usb 1-10.3.4.2: SerialNumber: 48:CA:43:54:22:24
[ 7167.442432] cdc_acm 1-10.3.4.2:1.0: ttyACM0: USB ACM device
[ 7171.301856] usb 1-10.3.4.2: USB disconnect, device number 46
```
 
journalctl output:
```
Jul 03 21:14:52 kernel: usb 1-10.3.4.2: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
Jul 03 21:14:52 kernel: usb 1-10.3.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jul 03 21:14:52 kernel: usb 1-10.3.4.2: Product: USB JTAG/serial debug unit
Jul 03 21:14:52 kernel: usb 1-10.3.4.2: Manufacturer: Espressif
Jul 03 21:14:52 kernel: usb 1-10.3.4.2: SerialNumber: 48:CA:43:54:22:24
Jul 03 21:14:52 kernel: cdc_acm 1-10.3.4.2:1.0: ttyACM0: USB ACM device
Jul 03 21:14:52 mtp-probe[44682]: checking bus 1, device 44: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10.3/1-10.3.4/1-10.3.4.2"
Jul 03 21:14:52 mtp-probe[44682]: bus: 1, device: 44 was not an MTP device
[... 9 lines cut ...]
Jul 03 21:14:57 kernel: usb 1-10.3.4.2: Product: USB JTAG/serial debug unit
Jul 03 21:14:57 kernel: usb 1-10.3.4.2: Manufacturer: Espressif
Jul 03 21:14:57 kernel: usb 1-10.3.4.2: SerialNumber: 48:CA:43:54:22:24
Jul 03 21:14:57 kernel: cdc_acm 1-10.3.4.2:1.0: ttyACM0: USB ACM device
Jul 03 21:14:57 mtp-probe[44730]: checking bus 1, device 46: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10.3/1-10.3.4/1-10.3.4.2"
Jul 03 21:14:57 mtp-probe[44730]: bus: 1, device: 46 was not an MTP device
Jul 03 21:14:57 mtp-probe[44732]: checking bus 1, device 46: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10.3/1-10.3.4/1-10.3.4.2"
Jul 03 21:14:57 mtp-probe[44732]: bus: 1, device: 46 was not an MTP device
Jul 03 21:15:02 kernel: usb 1-10.3.4.2: USB disconnect, device number 46
Jul 03 21:15:02 kernel: usb 1-10.3.4.2: new full-speed USB device number 47 using xhci_hcd
Jul 03 21:15:02 kernel: usb 1-10.3.4.2: device descriptor read/64, error -32
Jul 03 21:15:03 kernel: usb 1-10.3.4.2: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
Jul 03 21:15:03 kernel: usb 1-10.3.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[... cut ...]
Jul 03 21:15:03 mtp-probe[44778]: bus: 1, device: 47 was not an MTP device
Jul 03 21:15:07 kernel: usb 1-10.3.4.2: USB disconnect, device number 47
```

The one I just received still flickers between black/white when I change pages. It took a while to get it to connect to wifi (I had to use a hotspot on my phone). I haven't yet tried to connect it via USB, but I'll report back here after I do. I am also interested to find out if there's a straightforward way to manage the watch over USB (wifi config, firmware, watch faces, etc).

sqfmi, 2024-07-14:
@davidhaley are you following the instructions here? https://watchy.sqfmi.com/docs/getting-started/
You do not need to install any USB serial drivers

paolorussian, 2024-07-15:
> Hello, can you provide a link to the documentation you are using?

https://watchy.sqfmi.com/docs/getting-started/

paolorussian, 2024-07-15:
I've been fiddling with some incompatible libraries and it looks like we're getting somewhere now. Now I get /dewv/ttyACM0 enumerated but if I choose it then esptool returns: This chip is ESP32-S3 not ESP32. Wrong --chip argument?

paolorussian, 2024-07-15:
solved, it was a combinations of factors, mostly due that my first unit I've got had a broken back smd button, so I didn't bother pressing buttons to enter bootloader mode.
If someone else got a recent unit and is facing this issue, consider this:
1) it's necessary to press buttons in order to flash a firmware. It's not like a regular esp32 board where it mostly enters bootloader by itself. Unit with broken buttons = no flashing.
2) About watchy may tell your unit is v1.5 if your unit is recent,don't follow the instructions telling you to select board>watchy / version 1.5. Instead do this [https://github.com/sqfmi/Watchy/issues/252](url)

closed 2024-07-15 by paolorussian