sqfmi/Watchy #271: compile error wtachy V1.0 and arduino ide 2.3.3

kosh-mclien, 2024-10-31:
I had to update my arduino ide (for other reasons than watchy).

So I udated/reinstalled everything according to the manual (https://watchy.sqfmi.com/docs/legacy).

Every example watchface I try to upload results in this errors:
`WARNING: library Rtc_Pcf8563 claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
In file included from ~/.arduino15/packages/esp32/hardware/esp32/3.0.7/cores/esp32/esp32-hal-gpio.h:29,
                 from ~/.arduino15/packages/esp32/hardware/esp32/3.0.7/cores/esp32/esp32-hal.h:75,
                 from ~/.arduino15/packages/esp32/hardware/esp32/3.0.7/cores/esp32/Arduino.h:36,
                 from ~/Arduino/libraries/Watchy/src/Watchy.h:4,
                 from ~/Arduino/libraries/Watchy/src/Watchy.cpp:1:
~/Arduino/libraries/Watchy/src/Watchy.cpp: In member function 'void Watchy::deepSleep()':
~/.arduino15/packages/esp32/hardware/esp32/3.0.7/variants/watchy/pins_arduino.h:46:23: error: 'GPIO_SEL_26' was not declared in this scope; did you mean 'GPIO_NUM_26'?
   46 | #define MENU_BTN_MASK GPIO_SEL_26
      |                       ^~~~~~~~~~~
~/.arduino15/packages/esp32/hardware/esp32/3.0.7/variants/watchy/pins_arduino.h:50:23: note: in expansion of macro 'MENU_BTN_MASK'
   50 | #define BTN_PIN_MASK  MENU_BTN_MASK | BACK_BTN_MASK | UP_BTN_MASK | DOWN_BTN_MASK
      |                       ^~~~~~~~~~~~~
~/Arduino/libraries/Watchy/src/Watchy.cpp:129:7: note: in expansion of macro 'BTN_PIN_MASK'
  129 |       BTN_PIN_MASK,
      |       ^~~~~~~~~~~~
~/.arduino15/packages/esp32/hardware/esp32/3.0.7/variants/watchy/pins_arduino.h:47:23: error: 'GPIO_SEL_25' was not declared in this scope; did you mean 'GPIO_NUM_25'?
   47 | #define BACK_BTN_MASK GPIO_SEL_25
      |                       ^~~~~~~~~~~
~/.arduino15/packages/esp32/hardware/esp32/3.0.7/variants/watchy/pins_arduino.h:50:39: note: in expansion of macro 'BACK_BTN_MASK'
   50 | #define BTN_PIN_MASK  MENU_BTN_MASK | BACK_BTN_MASK | UP_BTN_MASK | DOWN_BTN_MASK
      |                                       ^~~~~~~~~~~~~
~/Arduino/libraries/Watchy/src/Watchy.cpp:129:7: note: in expansion of macro 'BTN_PIN_MASK'
  129 |       BTN_PIN_MASK,
      |       ^~~~~~~~~~~~
~/.arduino15/packages/esp32/hardware/esp32/3.0.7/variants/watchy/pins_arduino.h:32:21: error: 'GPIO_SEL_32' was not declared in this scope; did you mean 'GPIO_NUM_32'?
   32 | #define UP_BTN_MASK GPIO_SEL_32
      |                     ^~~~~~~~~~~
~/.arduino15/packages/esp32/hardware/esp32/3.0.7/variants/watchy/pins_arduino.h:50:55: note: in expansion of macro 'UP_BTN_MASK'
   50 | #define BTN_PIN_MASK  MENU_BTN_MASK | BACK_BTN_MASK | UP_BTN_MASK | DOWN_BTN_MASK
      |                                                       ^~~~~~~~~~~
~/Arduino/libraries/Watchy/src/Watchy.cpp:129:7: note: in expansion of macro 'BTN_PIN_MASK'
  129 |       BTN_PIN_MASK,
      |       ^~~~~~~~~~~~
~/.arduino15/packages/esp32/hardware/esp32/3.0.7/variants/watchy/pins_arduino.h:48:23: error: 'GPIO_SEL_4' was not declared in this scope; did you mean 'GPIO_NUM_4'?
   48 | #define DOWN_BTN_MASK GPIO_SEL_4
      |                       ^~~~~~~~~~
~/.arduino15/packages/esp32/hardware/esp32/3.0.7/variants/watchy/pins_arduino.h:50:69: note: in expansion of macro 'DOWN_BTN_MASK'
   50 | #define BTN_PIN_MASK  MENU_BTN_MASK | BACK_BTN_MASK | UP_BTN_MASK | DOWN_BTN_MASK
      |                                                                     ^~~~~~~~~~~~~
~/Arduino/libraries/Watchy/src/Watchy.cpp:129:7: note: in expansion of macro 'BTN_PIN_MASK'
  129 |       BTN_PIN_MASK,
      |       ^~~~~~~~~~~~
~/Arduino/libraries/Watchy/src/Watchy.cpp: In member function 'virtual void Watchy::handleButtonPress()':
~/.arduino15/packages/esp32/hardware/esp32/3.0.7/variants/watchy/pins_arduino.h:46:23: error: 'GPIO_SEL_26' was not declared in this scope; did you mean 'GPIO_NUM_26'?
   46 | #define MENU_BTN_MASK GPIO_SEL_26
      |                       ^~~~~~~~~~~
~/Arduino/libraries/Watchy/src/Watchy.cpp:138:19: note: in expansion of macro 'MENU_BTN_MASK'
  138 |   if (wakeupBit & MENU_BTN_MASK) {
      |                   ^~~~~~~~~~~~~
~/.arduino15/packages/esp32/hardware/esp32/3.0.7/variants/watchy/pins_arduino.h:47:23: error: 'GPIO_SEL_25' was not declared in this scope; did you mean 'GPIO_NUM_25'?
   47 | #define BACK_BTN_MASK GPIO_SEL_25
      |                       ^~~~~~~~~~~
~/Arduino/libraries/Watchy/src/Watchy.cpp:174:24: note: in expansion of macro 'BACK_BTN_MASK'
  174 |   else if (wakeupBit & BACK_BTN_MASK) {
      |                        ^~~~~~~~~~~~~
~/.arduino15/packages/esp32/hardware/esp32/3.0.7/variants/watchy/pins_arduino.h:32:21: error: 'GPIO_SEL_32' was not declared in this scope; did you mean 'GPIO_NUM_32'?
   32 | #define UP_BTN_MASK GPIO_SEL_32
      |                     ^~~~~~~~~~~
~/Arduino/libraries/Watchy/src/Watchy.cpp:187:24: note: in expansion of macro 'UP_BTN_MASK'
  187 |   else if (wakeupBit & UP_BTN_MASK) {
      |                        ^~~~~~~~~~~
~/.arduino15/packages/esp32/hardware/esp32/3.0.7/variants/watchy/pins_arduino.h:48:23: error: 'GPIO_SEL_4' was not declared in this scope; did you mean 'GPIO_NUM_4'?
   48 | #define DOWN_BTN_MASK GPIO_SEL_4
      |                       ^~~~~~~~~~
~/Arduino/libraries/Watchy/src/Watchy.cpp:199:24: note: in expansion of macro 'DOWN_BTN_MASK'
  199 |   else if (wakeupBit & DOWN_BTN_MASK) {
      |                        ^~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1`
so there seems to be something wrong with the definition of the pins in pins_arduino.h

Or I messed something up with the ide update (I have no idea why all the hardware packages got installed in ".arduino15")

holamlui, 2025-01-01:
I see you are using esp32 library 3.0.7? I had same issue on watchy v2.0, tried to downgrade esp32 library to 2.0.17 and it worked.

In the non-legacy official getting-started document step 4:
`Open Tools > Board > Boards Manager and install v2.0.17 of esp32 platform. Note: there are compatibility issues with v3.x.x`
[https://watchy.sqfmi.com/docs/getting-started](https://watchy.sqfmi.com/docs/getting-started)
Maybe it is same case for watchy v1.0

fdufnews, 2025-06-16:
In 3.0.x GPIO_SEL_x are not defined.
In the defines in pins_arduino.h:,  GPIO_SEL_x shall be replaced with (BIT64(x))