sqfmi/Watchy #277: Faulty board?

mjn101, 2025-01-18:
Just received a watchy from mouser uk. With the out of the box firmware the motor test isn’t working - I see the word “buzz”, but there is no buzz. 
Also in the menu the selection moves itself up every ~5 seconds without a button being pressed. Finally, every time a new screen is displayed, the display flickers from black on white to white on back (when going to the menu, any option on the menu, back to the watch face…). There is a flicker in between, not a straight change. 
Should I leave well alone and get it swapped, or is it okay to mess with the firmware to see if it gets better? Pardon my ignorance, for my sins I’m historically more of a windows sys admin guy than open source hardware tinkerer and am not a regular GitHub user!

sqfmi, 2025-02-02:
Hey @mjn101 have you reached out to Mouser for a replacement yet? The motor should buzz if you see the word 'buzz'.

As for the display going from black to white during refresh, that is the normal process for e-ink displays during full refresh. Check out this video to learn more: https://www.youtube.com/watch?v=FXIh4Ejwzaw

Once you receive your new Watchy, remember to update to the latest firmware. Let us know if you have any other issues, and join our Discord for community updates https://discord.gg/ZXDegGV8E7

Thanks!

codememoir, 2025-02-22:
Hi [sqfmi](https://github.com/sqfmi),  I also have same issue for watchy which i recently bought from mouser.ca, after assembly when i tried the motor test, the screen just display "buzz" without the actual motor buzzing. Is there a way to trouble-shoot this? Or this is a defective board as well?

Hi [mjn101](https://github.com/mjn101), we're you able to solve the issue with the buzzer?

Thanks and more power to you guys!

codememoir, 2025-02-24:
> Hi [@codememoir](https://github.com/codememoir), got to be honest, mouser offered me a refund and I took it (no slight to [@sqfmi](https://github.com/sqfmi), awesome kit, I just realise I don’t have the time to play with it at the moment)! All the best with your watchy :)

Thank you very much @mjn101 for sharing, really appreciated it!

codememoir, 2025-03-24:
Same outcome... Mouser also offered a refund and I took it as well.  Thanks to Mouser's excellent after sales customer service!

aagorchakov, 2026-04-09:
> the aggressive flickering between screen face changes

Flickering is not a problem, it's a full refresh, e-ink displays requires it to fully erase previous image. But you can disable it in the Watchy code. I disabled it for almost all use cases and it works fine.